imbushuo/mac-precision-touchpad #175: How to remove driver and certificate

mau-lopez, 2019-03-04:
Hi imbushuo, if it is not a problem could you please let me know the steps and locations to remove the driver and certificate? I'm waiting to the release of your driver to support the MacBook Pro 13" Retina Early 2015, I installed a couple versions of your driver before noticing that there was no support for my computer yet, but I like to keep my Win 10 installation as clean as possible. The installed drivers, although they don't give any new functionality,  seem to give no problems by the way. Thanks a lot in advance!

davidc14, 2019-03-12:
I'm also keen to find out how to uninstall this!

XTroyZe, 2019-03-16:
I don't know how to uninstall the certificate but to uninstall it (at least this is how I did it on my MBP early 2011):

1) Open up to device manager
2) Find Apple Precision Drivers (in my case it was under Human Interface Devices)
3) Right click it and click "update driver" then "browse my computer for driver software" then "let me pick from a list of available drivers on my computer"
4) There should be a list of drivers shown. Click on "Apple Multitouch" then click next.
5) It should install the original drivers and should be working by now.

NOTE: If you want to switch back to the Apple Precision Drivers, you can use the same approach and instead of selecting Apple Multitouch, choose Apple Precision Drivers.
Also, if for some reason there is no option to choose Apple Multitouch, I think you may be able to choose USB Input Device and then reinstall bootcamp drivers after this. You'll probably want to keep a mouse on hand because I think it breaks the touchpad for a little bit.

The driver is good, but the reason I decided to switch back was because the touchpad was not sensitive enough (ie: it required a little more pressure to operate the touchpad which was a little annoying) and I couldn't use the bottom left hand corner of the touchpad as a left click properly. Otherwise all the gestures work really nicely and it was great to have experienced them. At least with this method I can easily switch between the drivers.

mau-lopez, 2019-03-17:
Thanks a lot for your time and response XTroyZe, this is a very useful approach. Cheers

closed 2019-08-06 by imbushuo