imbushuo/mac-precision-touchpad #180: Conflict with MagicUtilities (& some general PC issues)

YetAnotherSN, 2019-04-21:
These are really more of a headsup for anyone struggling to get their MT2 to work than it is an issue with the drivers themselves. It would be helpful if tips like this were in the installation instructions, though.

When trying out the few different options for using the MT2 with a PC (Surface Pro 6), I installed MagicUtilites first. I then tried this driver. The "installation successful" message would pop up, but no drivers actually installed. Finally, I tried going back through older versions and found Drivers.2014 (univ-b1) to work. It was SUPER sluggish, though, to the point that it was unusable.

Long story short, I eventually realized there was still an active driver from MagicUtilites despite having uninstalled the program. After removing this, my MT2 is working great with the mac precision touchpad driver. Reinstalling MagicUtilities replicated the sluggishness when the MT2 is plugged in via USB, so I'm confident that was the issue. I was hoping to use MagicUtilities to have _some_ functionality while using bluetooth, but that's clearly not an option.

Regardless, thank you very much to everyone who has put in time on this project. This is the only option for an external high precision touchpad and I'm very grateful. I do hope someone is still working on getting BT to work :)

DrJedikiah, 2019-04-21:
Which built you have installed? I have MT2 and I'm struggling even moving the cursor

YetAnotherSN, 2019-04-21:
So, I know that to get it to actually instal, I had to use the [beta 1 .inf](https://github.com/imbushuo/mac-precision-touchpad/releases/tag/univ-b1)

But, I just went into Device Manager and looked at the driver, and it says the date is 02/28/18 and version 20.44.29.311, which is the [1802.M2 driver](https://github.com/imbushuo/mac-precision-touchpad/releases/tag/univ-b1-i3). Looking through the log, I must have updated it and forgot about it.

MagicUtilities, 2019-04-22:
Hey [Magic Utilities](https://magicutilities.net/) here, @YetAnotherSN: Our drivers should uninstall, we have no know issue, but a computer restart may sometimes required to remove the driver completely.

However, unfortunately this precision trackpad driver does **add** as a lower filter driver, but it should **replace** any lower filter driver (in the .inf change: HKR,,"LowerFilters",0x0001000**0**,"WUDFRd"), so the issue you described would not happen in the first place. Both driver can't coexist together as they modify the HID descriptor and the raw data for their own needs.

Anyhow, you should be able to use this driver on USB and the Magic Utilities driver on Bluetooth. You need to manually assign this driver on the USB connection in the Device Manager. [Contact us](https://magicutilities.net/contact) if you need detailed help.

DrJedikiah, 2019-04-22:
> Hey [Magic Utilities](https://magicutilities.net/) here, @YetAnotherSN: Our drivers should uninstall, we have no know issue, but a computer restart may sometimes required to remove the driver [...]

Why don't you guys just make a precision drivers that also works with bluetooth and all of us happy :D

MagicUtilities, 2019-04-22:
Well that's a good point and we really considered this for a long time. But if we would do this there would be no Windows 7 support, no 2 finger back/forward navigation, no 3 finger dragging, no accelerated scrolling, ... , basically we would give any touch control in any aspect away to Windows 10.

DrJedikiah, 2019-04-22:
Thanks for quick reply, or even replying :)

I'm using MagicUtilites for now but I would prefer Precision over the other features and I bet lot of people would as well. Only thing I can recommend is making two options like precision and the normal drivers and since Apple devices are not so frequently manufactured, device count will be limited and I guess this will make you guys job easier.

I also have a laptop that supports precision natively and I must add some info. You can set any gesture you want with precision drivers, it's not just the fluidity like as my config, I can change my volume with three finger up/down and it's so precise as good as a DJ pitch wheel. I can go back and forward on the web using three finger left/right. Three finger tap acts as middle click. Four finger down refreshes the page, four finger tap pauses the music etc. etc.

This is what I made with the precision drivers and I fell in love with them. I'm really hoping to buy precision version if you guys make it happen even if it's more expensive. I mean people can choose according to operating system they use and I don't think people will downgrade their system often.

YetAnotherSN, 2019-04-22:
> basically we would give any touch control in any aspect away to Windows 10.

That is exactly what I am looking for. Essentially, I'm moving to a trackpad and willing to pay several times the price of other touchpads for pinch to zoom (not ctrl+/-, but true pinch to zoom) and smooth multidirectional scrolling. Almost any other option I can assign using the "Advanced Gestures" native to Windows 10.

TheAussieStew, 2019-04-25:
Likewise, I'd be happy to donate/pay for a precision version of the MT2 drivers that works with Bluetooth and not just USB. I quite like how Windows handles precision trackpads and I'd be happy to cede control in that respect.

freecodecampster, 2019-05-28:
> So, I know that to get it to actually instal, I had to use the beta 1 .inf
> But, I just went into Device Manager and looked at the driver, and it says the date is 02/28/18 and version 20.44.29.311, which is the 1802.M2 driver. Looking through the log, I must have updated it and forgot about it.

I'm running Windows 10 bootcamp on my iMac late 2013 with a Magic Trackpad 2. I tried most of the drivers from https://github.com/imbushuo/mac-precision-touchpad/releases. Multi-touch would work but the cursor would freeze often. I didn't try 1802.M2 Interim Release as I had tried 1806 until I read your post. With 1802 installed the cursor is working and multi-touch so thanks!