imbushuo/mac-precision-touchpad #183: Code 52, certificates installed to machine roots

nvella, 2019-04-25:
Hi again,

I've added the certificate to the machine Trusted Root Certification Authorities and Trusted Publisher stores. Installing the driver either via Device Manager or right-clicking and selecting install seem to both yield a Code 52 "Windows cannot verify the digital signatures for the driver required" error. Just wondering if I'm missing anything? Windows 10 1903 and MacBook Pro 13" Early 2015.

Kind regards,
Nick

imbushuo, 2019-07-21:
Resolved as today

closed 2019-07-21 by imbushuo

phyllisstein, 2020-01-13:
I just ran into this issue following a clean installation of Windows 10 on a 2019 MacBook Pro. I used the new setup process, which seemed to work fine, but after rebooting there was no cursor on the screen and Device Manager was spitting out this error.

[image]

imbushuo, 2020-01-13:
T2 devices need to turn of Secure Boot explicitly as of now. Try to redo the installation again with stale certificates removed from the machine.

Thanks,
Bingxing Wang

Get Outlook<https://aka.ms/qtex0l> for iOS

[IMG_0039]<https://user-images.githubusercontent.com/2207980/72236013-1c5cb300-35a3-11ea-83a3-7e704cd988f0.jpeg>

phyllisstein, 2020-01-19:
That worked beautifully, thanks so much.