imbushuo/mac-precision-touchpad #184: Does not work for Macbook Pro 2018 13" no touchbar

JByfordRew, 2019-05-03:
2018 Macbook Pro 13"

Trying a number of means to update the trackpad driver but does not seem to work.  Getting this.

[image]

imbushuo, 2019-10-21:
Could you provide the SPI trackpad device ID?

HalfbyteHeroes, 2019-10-26:
I've got the same issue with my 2016 13 inch model with Touch Bar. How would I find out what the device ID is?

imbushuo, 2019-10-26:
@HalfbyteHeroes I would blindly assume the device be located under "Apple SPI Device" if you view the Device Manager in connection topology. Otherwise, try to find a "Apple Multi-touch" device, open its properties and check the hardware device ID.

Try installing the latest release first, this one eliminates most confusion in connection technologies and model identification.

HalfbyteHeroes, 2019-10-30:
Installing the latest release is what I tried first, but the installation failed with the error message already posted in the first comment. I've also manually installed the certificate, but that didn't help either. Manually installing the SPI driver from the previous release worked, but I couldn't use that driver with the `Apple SPI Device`. Anyway, here are the hardware IDs:
```
ACPI\VEN_APP&DEV_000D
ACPI\APP000D
*APP000D
```

Let me know if you need any more info.

imbushuo, 2019-10-30:
@HalfbyteHeroes APP000D is not the trackpad device. There are three Apple SPI Device which remains unused on MacBook (they are Force Touch components) and you are expected installation failures on those.

You can find either Apple Multi-Touch or some input device under Apple SPI Device (actually a bus device).

imbushuo, 2019-10-30:
[image]

An example of the topology FYI.

HalfbyteHeroes, 2019-10-30:
Ahh - I've never used that topology view before. Those are the two devices listed under `Apple SPI Trackpad`:

[image]
[image]

HalfbyteHeroes, 2019-10-30:
Yes, that did indeed work! Thank you very much! 🙂

closed 2019-10-30 by imbushuo