imbushuo/mac-precision-touchpad #185: Support for Magic Mouse

DarrellDD, 2019-05-05:
Would it be interesting to ad support to the magic mouse aswell? In terms of smooth scrolling and multitouch. As perhaps a mouse/trackpad hybrid? I'm already really pleased with the work done to the magic trackpad 2, and would like to put my magic mouse to good use too.

JLIspace, 2019-07-08:
I'm using a first gen magic mouse with the apple bootcamp driver and scrolling works fine.

DarrellDD, 2019-07-09:
It's not smooth though. Your windows mouse scroll option is being changed aswell which leads to inconsistent scrolling between pointing devices. The Precision Touch prevents this.

imbushuo, 2019-07-21:
It might work, but using Windows gestures on an arc surface just doesn't make sense to me. Currently I have no plan for supporting Magic Mouse family.

closed 2019-07-21 by imbushuo