imbushuo/mac-precision-touchpad #187: model support MacBook Air (Retina, 13-inch, 2018)

kylus, 2019-05-07:
MacBookAir8,1

Product ID：	0x027a
Vendor ID：	0x05ac（Apple Inc.）
ST Version：	9.13

kylus, 2019-05-07:
[image]

imbushuo, 2019-07-30:
Supported as of today.

closed 2019-07-30 by imbushuo

liustar2008, 2019-07-30:
The latest driver doesn't work on the machine.
MacBookAir8,1

Product ID：	0x027a
Vendor ID：	0x05ac（Apple Inc.）
ST Version：	9.13

The driver only support the 0x027d, not 0x027a. please check the driver. Thanks!

kylus, 2019-08-05:
same problem as liustar2008,
driver can't work on my machine

Product ID：	0x027a
Vendor ID：	0x05ac（Apple Inc.）
ST Version：	9.13

kylus, 2019-08-05:
> Supported as of today.
same problem as liustar2008,
driver can't work on my machine

Product ID：	0x027a
Vendor ID：	0x05ac（Apple Inc.）
ST Version：	9.13

@imbushuo