imbushuo/mac-precision-touchpad #190: MacBookPro 15,1 (2019, Radeon Pro 560X, Core i9-9980H) USB Device ID's

Uhucream, 2019-06-05:
[image]

[image]

Uhucream, 2019-06-05:
[image]

There is no SPI devices on MacBook Pro 2019.

imbushuo, 2019-07-21:
I have introduced support for this model in https://github.com/imbushuo/mac-precision-touchpad/releases/tag/1907-3529. However, since some calibration data are carried from 13-inch, it may not work optimally. Please try and provide feedback!

closed 2019-07-29 by imbushuo

Uhucream, 2019-09-30:
So sorry to be late for responding.
I had installed this driver, it works.
But out range of 13-inch's trackpad couldn't be recognized in my MBP 15-inch.

imbushuo, 2019-09-30:
@Uhucream Okay so I know that the 15inch have larger space than the 13inch one - will work on a fix for that.

Uhucream, 2019-10-10:
I see.
By the way, I measure the trackpad size of 15-inch MacBook Pro.
Its size is about 10cm x 15cm.

[image]
[image]