imbushuo/mac-precision-touchpad #193: MODEL SUPPORT REQUEST For MacBookPro15,2 (13" 2018 with Touchbar)

rbdej, 2019-06-16:
Hi, I'd like to submit my Macbook's details. Hope this helps on your work :)
P.S. I didn't find the Driver ID under SPI (mine's empty), I found it under the USB section in System Report app.

Model Name:	MacBook Pro 13" 2018 with Touch Bar
Model Identifier:	MacBookPro15,2
Processor Name:	Intel Core i5
Processor Speed:	2.3 GHz
Memory:	8 GB 2133 MHz LPDDR3
Graphics:	Intel Iris Plus Graphics 655 1536 MB

Apple Internal Keyboard / Trackpad:
  Product ID:	0x027b
  Vendor ID:	0x05ac (Apple Inc.)
  Version:	9.27
  Serial Number:	FM7827301XGJ3VXAS+WRN
  Speed:	Up to 480 Mb/sec
  Manufacturer:	Apple Inc.
  Location ID:	0x80500000 / 3
  Current Available (mA):	500
  Current Required (mA):	500
  Extra Operating Current (mA):	0
  Built-In:	Yes

Screenshots:
[image]
[image]

Uhucream, 2019-06-21:
I agree with you.
I can't use Windows 10 on BootCamp without it.

imbushuo, 2019-07-19:
Work started on https://github.com/imbushuo/mac-precision-touchpad/tree/t2, you can keep an eye on it.

rbdej, 2019-07-25:
Thank you imbushuo for the great work 😄. I'll go test this out around next month.

imbushuo, 2019-07-29:
Supported as https://github.com/imbushuo/mac-precision-touchpad/releases/tag/1907-3539.

closed 2019-07-29 by imbushuo