imbushuo/mac-precision-touchpad #201: Build 3524 Digital Signature Issue

mau-lopez, 2019-07-20:
On MBP 12,1 Windows 10 May 2019 Update, when trying to install the INF file under C:\Users\user\Downloads\Drivers\AmtPtpDeviceSpiKm\x64\ReleaseSigned, error message: 

Third party INF does not contain digital signature information

Please advise.

brandon-james105, 2019-07-20:
Hey. You need to restart Windows in a mode that doesn't check unsigned drivers. To do this, you go to the Start Menu, click power > restart and click restart with the shift key pressed down. When it restarts it gives you options to go to an advanced startup and lists one an option for not checking unsigned drivers. So, I managed to get that to work that way. The driver starts in that mode. However, when I restarted and that mode was disabled my trackpad no longer worked. Here's the message I have now:
[image]
I would need some help with that. I'm pretty sure I installed the security cert to the root store.

imbushuo, 2019-07-21:
@brandon-james105 Can you confirm if this certificate is located in **Trusted Publisher**? According to Microsoft documentation this might be necessary to make driver load. During INF installation, the dialog that ask for publisher confirmation should completes this but I have not validated the upgrade path.

In other words, normal cross-certificate works for kernel-mode drivers unless Secure Boot is enabled. Secure Boot does not present on non-T2 MacBooks so it should work. On T2 devices, Secure Boot needs to be manually disabled via Recovery Utility.

imbushuo, 2019-07-21:
@mau-lopez Can confirm there is a certificate configuration issue on my "Golden Machine" (Build agent). Investigation is in progress.

brandon-james105, 2019-07-21:
@imbushuo I just installed the driver again and manually selected Trusted Publisher. A note - I never was prompted for the cert when I installed the INF.

brandon-james105, 2019-07-21:
When I go through the "Have Disk" dialog box as so -
[image]
I get 
[image]

imbushuo, 2019-07-21:
@brandon-james105 @mau-lopez It's a CI-side issue, it will take some to fix once I go home tonight (I am in Disneyland now)

brandon-james105, 2019-07-21:
@imbushuo Hey man, enjoy yourself. You're doing more than I would have had I been at a theme park. I woulda ignored support requests!

I'm actually two hours away from Disney World where I live now. :) I'm real close to Orlando.

imbushuo, 2019-07-21:
@brandon-james105 @mau-lopez Check https://github.com/imbushuo/mac-precision-touchpad/releases/tag/1907-3529, which should addressed the issue.

I am actually at the California one :) So far away from the one in Orlando :P

closed 2019-07-21 by imbushuo

brandon-james105, 2019-07-21:
Thanks so much Ben. My trackpad experience is so much more enjoyable now! I'm glad this was something that could be accomplished. Wish I could help but I'm a lowly web/apps dev. lol. Perhaps if I were to volunteer to make a pretty github.io download page so it indexes in search engines eand this driver is easier to find?

By the way I think the last time we chatted I had a 2013 MBP. I upgraded to the 2016 MBP last year when I ran out of space on the 2013.

brandon-james105, 2019-07-21:
Thanks so much Ben. My trackpad experience is so much more enjoyable now! I'm glad this was something that could be accomplished. Wish I could help but I'm a lowly web/apps dev. lol. Perhaps if I were to volunteer to make a pretty github.io download page so it indexes in search engines and this driver is easier to find?

By the way I think the last time we chatted I had a 2013 MBP. I upgraded to the 2016 MBP last year when I ran out of space on the 2013.

mau-lopez, 2019-07-23:
Hi Ben, guys, has this issue been solved? I kind of lost track, but can see that this ticket is closed. If indeed it's been solved, please advise on how to proceed. Thanks a lot!

imbushuo, 2019-07-23:
@mau-lopez Install build 3529+ and proceed installation as described in README as well as https://github.com/imbushuo/mac-precision-touchpad/issues/201#issuecomment-513507866.

mau-lopez, 2019-07-23:
Thanks a lot. Yes, I can confirm that by right clicking the certificate file and choosing install and then choosing to install in the "Trusted Publishers" folder, now this amazing driver is working !! Yeah !! Thanks a lot for this driver Ben !! Fantastic