imbushuo/mac-precision-touchpad #203: Trackpad doesn't work after sleep

vsegdacocacola, 2019-07-23:
Laptop: Macbook Pro 15 Late 2016 (A1707)
OS: Windows 10 Pro 1903 (Build 18362.239)
Driver version: 20.52.33.805 (20/07/2019)

After awake from sleep tochpad is irresponsible. Device manager gives an error for Apple SPI Precision Touchpad Device "Windows has stopped this device because it has reported problems. (Code 43)" and device is marked with exclamation mark.

In Windows System logs there is message:

```
The system has returned from a low power state.
Sleep Time: ‎2019‎-‎07‎-‎23T08:33:53.621268000Z
Wake Time: ‎2019‎-‎07‎-‎23T10:11:24.829314100Z
Wake Source: Device -Apple SPI Device

```
No any other errors identified related to this device.

Walkarounds:
1. Touchpad works normally after reboot
2. Disabling and then enabling Apple SPI Precision Touchpad Device fix the issue. Can be automated using bat script and DevCon,exe from Windowd Driver Kit.

vsegdacocacola, 2019-07-23:
Regarding workaround 2:
1. Install devcon.exe from Windows Driver Kit
2. Find ID for your Apple SPI Precision Touchpad Device: 
`devcon.exe find *` 
3. Create batch script, which disable and then enable device
devcon.exe disable "<your device ID before second slash>"
devcon.exe enable "<your device ID before second slash>"
4. Create task with task scheduler to run batch trigger by "On workstation unlock"

ravenxiao454, 2019-07-24:
add an description:
This issue will happen when you wake by **pressing keybord/trackpad(except power button)**
However,if you **press power button** to wake your macbook,the trackpad work still normal

nvella, 2019-07-24:
Can also confirm this issue on 2015 13" MacBook Pro MacBookPro12,1. Didn't find anything unusual in eventvwr, "Windows has stopped this device because it has reported problems. (Code 43)" under Apple SPI Precision Trackpad Device in Device Manager.

Kent-Narun, 2019-07-24:
> 
> 
> add an description:
> This issue will happen when you wake by **pressing keybord/trackpad(except power button)**
> However,if you **press power button** to wake your macbook,the trackpad work still normal
> Laptop: Macbook pro 13' 2015
Yes it's the same for Me
Macbook Pro 13,2, Touch Bar 2016.

mau-lopez, 2019-07-24:
I can also confirm this issue on 2015 13" MacBook Pro MacBookPro12,1 exactly as nvella

TriHardStudios, 2019-07-25:
This is an example of a script. What @vsegdacocacola didn't work for me.

`@echo off
REM disables and reenables trackpad
devcon disable "@SPI\VID_05AC&PID_0276&MI_02\5&33D07321&0&02"

devcon enable "@SPI\VID_05AC&PID_0276&MI_02\5&33D07321&0&02"`

You need to replace the `"@SPI\*\*"` with @ what you find with `devcon list *` You need the entire string

enricmarques, 2019-08-04:
I can confirm this issue on two different laptops: a MacBook 12" 2017 and a MacBook Pro 13" 2017.

enricmarques, 2019-08-08:
I still have this issue with the latest update, Here is an screenshot after waking the laptop:

[image]

[image]

[image]

The laptop has been woken up automatically as soon as I plugged the power adapter, as it seems recent Mac computers wake when they detect a change in the AC adapter. So I haven't pressed any key.

Disabling and re-enabling the device fixes the issue.

lawiparadise, 2019-08-17:
I got same problem..

AuroraWright, 2019-08-28:
@imbushuo 
[LogSession_082819_234006.etl.zip](https://github.com/imbushuo/mac-precision-touchpad/files/3553021/LogSession_082819_234006.etl.zip)
From my 15" MBP 2016, I suspended and resumed twice before shutting down and trackpad failed to resume the first time

imbushuo, 2019-08-28:
@AuroraWright Try the driver build in https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_build/results?buildId=3567&view=artifacts and see if the situation improves. I will not sign off an official build in Github yet.

AuroraWright, 2019-08-29:
Still doesn't work. Not sure if this can be useful to you but I traced both cases. If you wake up using the power button, it wakes up correctly. If you wake up using keyboard/trackpad it doesn't work.
[Logsessions.zip](https://github.com/imbushuo/mac-precision-touchpad/files/3553076/Logsessions.zip)

cor, 2019-09-01:
I'm having the same issue on my 15" MacBook Pro 2017.

JohnnySun, 2019-09-02:
I‘m having the same issue on my 13 MacBook Pro 2017

imbushuo, 2019-09-02:
Might resolve with https://github.com/imbushuo/mac-precision-touchpad/commit/795e788be9a26354519e41bb4841dc6e2d5543e6; I am testing it locally for a while. Weirdly enough, it is hard to reproduce this on MacBook 12-inch.

Build artifacts: https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_build/results?buildId=3568

imbushuo, 2019-09-03:
Update: if you encountered BSoD with https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_build/results?buildId=3570, I would like to know your Boot Camp's driver version. It seems like an Apple upstream driver problem now :\

Kent-Narun, 2019-09-05:
> 
> 
> Update: if you encountered BSoD with https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_build/results?buildId=3570, I would like to know your Boot Camp's driver version. It seems like an Apple upstream driver problem now :\

I encountered BSoD while using build id 3570. **BootCamp Version 6.1**.

JohnnySun, 2019-09-06:
> Update: if you encountered BSoD with https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_build/results?buildId=3570, I would like to know your Boot Camp's driver version. It seems like an Apple upstream driver problem now :\

use this driver, sometime when open my mac, windows will crash. just got blue screen

xeijin, 2019-09-10:
Same thing for me, no mouse on wake from sleep.

Except I can’t even disable and re-enable the device — it says ‘the driver is not installed’.  When I try and either disable/re-enable, or manually point it to the drivers again it tells me I need to restart :(

The workaround mentioned to wake the laptop with the power button does not work for me.

Late 2016, 13” MacBook Pro with Touch Bar

Goyadoy, 2019-09-30:
I have noticet another interesting behaviour on my MacBook Pro 15 2017.
1. Go to sleep mode, then touch the touchpad, then press Enter to wake up - I have no pointer in windows, driver needs to be reenabled.
2. Go to sleep mode, do not touch the touchpad, press Enter - drivers works just fine.
3. And if I press the touchpad when macbook is in the sleep mode - I fell responce from the taptic engine, so it looks like touchpad is never fully sleeping.

orelh, 2019-10-11:
I'm experiencing the same issue. 2016 15" MBP with SPI drivers.

kalatehbali, 2019-10-20:
Same problem on: MBP 2017 - 15"

imbushuo, 2019-10-21:
Need someone to test https://ligstd.visualstudio.com/7694e0d0-94e3-4fd2-b39a-ecd261e1ba2e/_build/results?buildId=3662. I will monitor my MacBook 12-inch for a while.

imbushuo, 2019-10-22:
I've tried randomly wake the laptop up and put it into sleep (also hibernate) for a few times throughout the day. So far my trackpad doesn't die. However, I need more test data since the reproduce rate on MacBook8,1 / MacBook9,1 is significantly lower than other models.

imbushuo, 2019-10-27:
@kalatehbali The current situation is a bit confusing to me, can I confirm that you are on the latest release (or simply try the latest release in Github release page)?

kalatehbali, 2019-10-27:
> @kalatehbali The current situation is a bit confusing to me, can I confirm that you are on the latest release (or simply try the latest release in Github release page)?

Ok, I was sure that I installed the latest version, but I did it again with Build: 3665.
Meanwhile, I found another bug, which I think it's related to this.

So first, I still have this sleep issue whenever I wake up the MBP with the trackpad.
Recently when I want to log in into windows after a reset, my keyboard freeze, and nothing works except touch-bar. So in this situation, I only can shut-down the MBP by holding the touch-id button to make a hard reset.
If I try to quickly log-in into windows after seeing the welcome windows, all I can do is write the password and hit the enter to log-in, but after I see the desktop, my cursor becomes a blue circle and everything freeze.

imbushuo, 2019-10-27:
@kalatehbali The Apple SPI device is a bit weird as the stall of a single child device can affect others. What currently certain is that the problem is mitigated for a few other models, and I currently have no idea about the MBP’s case. Maybe I can get a device for test, but the lead time remains unpredictable.

Kent-Narun, 2019-10-31:
I hope this BSD related files for Macbook Pro 2016 512GB be helpful.

[Kernel_9f_cd257dae1ff721a157529411b5ae814c7cb4366_00000000_cab_32a32b3a-dac5-4069-8d52-25d972900859.zip](https://github.com/imbushuo/mac-precision-touchpad/files/3791957/Kernel_9f_cd257dae1ff721a157529411b5ae814c7cb4366_00000000_cab_32a32b3a-dac5-4069-8d52-25d972900859.zip)
[Kernel_9f_6455579375eb6653d7f61794867f9954efbdee_00000000_cab_7c91ac2c-7fab-449a-a3ee-3cb9e4d643ef.zip](https://github.com/imbushuo/mac-precision-touchpad/files/3791958/Kernel_9f_6455579375eb6653d7f61794867f9954efbdee_00000000_cab_7c91ac2c-7fab-449a-a3ee-3cb9e4d643ef.zip)

rdvm, 2019-10-31:
I have a MacBookPro 13,3 and was experiencing this issue with previous releases. But I've been using the driver version 4.4.30.422 for about a week and haven't had an unresponsive trackpad after sleep in that time.

I have encountered the same thing as @kalatehbali, though, with there being times when only the touchbar works on the lockscreen, and I have to power off the laptop.

Let me know if there is any information I can provide that would be helpful.

xeijin, 2019-11-04:
@rdvm as a workaround you might be able to kill `bootcamp.exe` in task manager, then run it again from the start menu  instead of re-starting the computr (I dont have this specific issue, but I've had to force-restart the touchbar before)

bhermann17, 2019-11-23:
Using some of the ideas above, I finally found an easy solution:
1. Open Windows PowerShell ISE as administrator
2. Run the following command to find the InstanceId of your Touchpad

Get-PnpDevice -FriendlyName "*Apple SPI Precision Touchpad Device*" | ft -wrap -AutoSize instanceid

3. Copy the value returned to you
4. Create a batch script that will disable and re-enable the Touchpad
5. Paste the following into the script replacing "INSTANCEID" with the value you copied

Disable-PnpDevice -InstanceId "INSTANCEID" -Confirm:$false
Enable-PnpDevice -InstanceId "INSTANCEID" -Confirm:$false

6. Create a task with Task Scheduler to run the script you just made
7. Create a trigger for the task using "on workstation unlock"

You're all set. Enjoy!

TP328, 2020-03-14:
Hello I am having troubles doing this step. Can someone take a screenshot on how to do step (5,6&7).
Thank you.

3. Copy the value returned to you
4. Create a batch script that will disable and re-enable the Touchpad
5. Paste the following into the script replacing "INSTANCEID" with the value you copied

Disable-PnpDevice -InstanceId "INSTANCEID" -Confirm:$false
Enable-PnpDevice -InstanceId "INSTANCEID" -Confirm:$false

6. Create a task with Task Scheduler to run the script you just made
7. Create a trigger for the task using "on workstation unlock"

Leeable, 2020-05-14:
1.make sure you already install visual studio、Microsoft Framework、Microsoft visual c++
2. search ‘devcon.installer’ in the github and download it， run it with admin and **"Add DevCon to environment path"** (inside verson is devcon 1709,but it works in 1909)
3.  Device Manager--Apple SPI Precision Touchpad Device---check the **hardware ID** (like 
                 SPI\VID_05ac&PID_0276&MI_02                )
4.open .txt  and write this and **cover your Hardware id with it**

@echo off
REM disables and reenables trackpad
devcon disable "SPI\VID_05ac&PID_0276&MI_02"

devcon enable "SPI\VID_05ac&PID_0276&MI_02"

5. rename it with .bat
6.Create a task with Task Scheduler to run the script you just made
7.Create a trigger for the task using "on workstation unlock"

Goyadoy, 2020-05-14:
Also I had issues with MacBook unable to wake from sleep (it crashed with BSOD about 1 of 5 times after I wake it). I solved it by adding bat file with following line in task scheduler with "on workstation lock" trigger. I think it can be very useful for a lot of people.

devcon.exe disable "SPI\VID_05AC&PID_0278&MI_02"

RDomak, 2020-06-06:
Can somebody explain for an amateur, how to solve this problem? Leeable wrote a nice instruction but I have no idea what are the steps to take exactly :) I am stuck at installing devcon, I guess. 

- I installed Visual Studio (but I have no idea, what it really is and how it works) 
- I guess / hope it has the mentioned modules "Microsoft Framework、Microsoft visual c++" 
- I downloaded ‘devcon.installer’ from the github (very proud of myself already for this :) and then I'm stuck. What does the following thing mean? 
 "run it with admin and "Add DevCon to environment path" (inside verson is devcon 1709,but it works in 1909)" 

Additionally I managed to find the hardware ID, created the .bat file also but it doesn't do anything as probably devcon is still not running.

RDomak, 2020-06-07:
Alright, I managed to install Devcon with the help of an excellent Github program: 
https://github.com/Drawbackz/DevCon-Installer/releases

I confirm that devcon.exe works now. 

BUT I still have a problem with disabling / enabling the device ID. 

When I run the devcon.exe with the correct ID number then I still get this kind of error: 

C:\Windows\system32>devcon disable "@SPI\VID_05ac&PID_0279&MI_02\5&33d07321&0&02"
SPI\VID_05AC&PID_0279&MI_02\5&33D07321&0&02                 : Disable failed
No matching devices found.

As you can see, the ID number is correct, I took it from the device properties. Also the syntax seems right for me. What can be the issue?

[image]

kalatehbali, 2020-06-08:
> Alright, I managed to install Devcon with the help of an excellent Github program:
> https://github.com/Drawbackz/DevCon-Installer/releases
>
> I confirm that devcon.exe works now.
>
> BUT I still have a [...]

Try to copy paste this below code and let me know if you can disable it or not:

`C:\Windows\system32>devcon disable @SPI\VID_05ac&PID_0279&MI_02\5&33d07321&0&02 `

RDomak, 2020-06-08:
Thank you very much, Kalatehibali, I tried but apparently it's not the right syntax. I tried both by adding the path and also without the path and just starting with devcon but I got this below error: 

[image]

kalatehbali, 2020-06-08:
> Thank you very much, Kalatehibali, I tried but apparently it's not the right syntax. I tried both by adding the path and also without the path and just starting with devcon but I got this below error:
> 
> [image]
Copy/Paset this please:

`devcon.exe disable "SPI\VID_05ac&PID_0279&MI_02"`

RDomak, 2020-06-09:
Thank you, it didn't work this way as well :( 

FInally I gave up and installed back the original Bootcamp driver. I also found a nice program called XMouse Button Control for Windows and it can manage the control quite nicely over the original touchpad too.  

https://www.highrez.co.uk/downloads/XMouseButtonControl.htm

Also I wanted to have the external mouse to scroll one way, and the touchpad the other way so the scroll reverse function by itself didn't help at this program. 

I had to change the scroll direction of the touchpad in the Windows Registry - just find the touchpad device and change the FlipFlopWheel value from 0 to 1.

RDomak, 2020-06-11:
Alright, back to square one: the original Bootcamp touchpad driver still sucks. It's just awful. It doesn't have a decent palm rejection so often when I type, it just suddenly deletes a chunk of my text. 

So I put back again the MS Precision driver yesterday and today I faced a new problem: after sleep when I turned on the computer it was frozen for a while (before login) and then I got a blue screen of death with the error code "Driver Power State Failure" 

Any good ideas?

ikappas, 2020-07-13:
This issue with the BSOD occurs on my Mac Book Pro 2016 w/ touchbar

TzuYangLin, 2020-07-29:
try 

devcon disable "SPI\VID_05ac&PID_0279&MI_02"

tigersfa8, 2020-09-10:
I'm not sure if this would need it's own issue log or not, but I'm using a KVM switch for two computers, and I have to unplug and plug the usb connection back in for the touchpad to be activated. I've tested this with a regular mouse, and no such issue exists.

jamesy0ung, 2020-09-11:
Same issue, Windows 10 Enterprise 2004 and MacBookPro 14,3

Bananaman, 2020-09-11:
I don't use a bootcamp mac but I read this whole thread out of interest. It sounds to me like Apple's bootcamp touchpad communication driver is buggy. And since imbushuo's hooks into Apple's driver to read the touchpad, and "transform it into a precision touchpad", it seems like the problems are in Apple's code. He said that Apple's bootcamp driver is very inconsistent and unreliable when returning from sleep.

bxff, 2021-05-12:
I am having the same issue but my keyboard also freezes, so the only way to get back control is to force reboot. Is there any fix for this yet? Also I don't get any problem if I boot with just taping the power button.

imbushuo, 2021-05-30:
Fixed as of today using a workaround. The defect is that you might have to wait for 5 seconds from wakeup to have a fully functional touchpad.

closed 2021-05-30 by imbushuo

bxff, 2021-06-01:
@imbushuo, Don't get me wrong, but does this also fix the problem where some people also have a frozen keyboard, whose power-off key only works?

imbushuo, 2021-06-01:
Yes they are the same thing, so this fix should also applicable to that scenario.

commit 41b8e67a (2019-08-28): SPI: Add retries for D0 enter/exit

Might resolve #203 and #99 but I am unable to reproduce the bug.
commit e92581b3 (2021-05-30): SPI: workaround parent device initialization delay from S3 wakeup

Fixes #203.