imbushuo/mac-precision-touchpad #204: Request for MacBook Air (Retina, 13-inch, 2019) T2

uueaaaa, 2019-07-24:
[image]

imbushuo, 2019-07-24:
@uueaaaa Can you provide physical dimension (width and length) of the new MacBook Air trackpad?

uueaaaa, 2019-07-24:
@imbushuo Thank you. width:120.5mm (or120mm?) length:82mm
[image]
[image]

imbushuo, 2019-07-29:
Supported as https://github.com/imbushuo/mac-precision-touchpad/releases/tag/1907-3539.

closed 2019-07-29 by imbushuo