imbushuo/mac-precision-touchpad #205: Touchpad stops working after installation of driver, Code 10 (2018 13")

dcai777, 2019-07-24:
Essentially what the title says, after installing the inf and using update driver and pointing to the inf in Device Manager it says "This device cannot start (Code 10)". One thing to note is that the device/hardware ID of the 2018 seems to be different than the one in your driver (presumably built around the 2019 IDs). In your driver, the trackpad is defined as USB\Vid_05ac&Pid027c&MI_02 and USB\Vid_05ac&Pid027d&MI_02 but the ID for the 2018 is USB\Vid_05ac&Pid027b&MI_02 (b instead of c/d).

imbushuo, 2019-07-24:
[image]

@dcai777 Each model requires both INF entry and a metadata entry. It is normal that you receive an error 10 because 2018 models have not yet been registered into the registry. I expect to add support for 2018 models shortly.

In additional, tracking model support using Github issues is not really effective, I am seeking alternate approach for this.

rbdej, 2019-07-29:
Im hoping for support for MacBookPro15,2 (2018 13") too.

imbushuo, 2019-07-29:
Supported as of today

closed 2019-07-29 by imbushuo