imbushuo/mac-precision-touchpad #206: Hold right-click unsupported (T2 MBP)? Prevents Citra Motion Controls Emulation

goldmastersims, 2019-07-25:
Hello,

Just recently installed the new precision drivers on a 2018 15'' MacBook Pro.

Holding right-click with the trackpad to emulate motion controls in the Citra 3DS Emulator still does not work unlike Citra in macOS. Both on Canary and Nightly versions.

Here is the specific scenario attempting to pass in the emulation with the precision touchpad drivers:
1. Stage 1-2 in emulated Kirby: Triple Deluxe.
2. The obstacle in the below screenshot on native 3DS hardware would require you to tilt the device.
3. You hold right click to emulate the 3DS' gyro movement.
4. This input is not registered, regardless of Apple's Boot Camp drivers or these drivers.

[image]

It seems hold right-click is not yet perfected this build.

I have been tracing down this issue for several months and found that it could only be a driver issue.

In Boot Camp with a MacBook Pro, the only way I can get through this is to connect a mouse via Bluetooth or one of the Thunderbolt 3 ports.

However, other basic functionality such as the scrolling works fine even when clicking while typing.

Am using the latest release for Apple T2 devices as of the time of posting.

imbushuo, 2019-07-25:
The PTP driver uses some HID collections that might be too modern to the emulator. Additionally, legacy input device such as mouse input is not implemented (the mouse you see in Windows is simulated by Windows itself). If you have other PTP devices such as Dell XPS 13 or Surface Pro, please also try it and send feedback.

encombhat, 2019-09-08:
Hold right-click is not working for Minecraft as well.

Bananaman, 2020-09-11:
This MS article seems related:

https://docs.microsoft.com/en-us/windows-hardware/design/component-guidelines/windows-precision-touchpad-required-hid-top-level-collections#input-mode-feature-report

> The input mode feature report is communicated by the host to the Windows Precision Touchpad to indicate which top-level collection should be used for input reporting. There are two collections which may be used for input reporting: the mouse collection and the Windows Precision Touchpad collection.
> 
> By default, upon cold-boot/power cycle, Windows Precision Touchpads shall report input by using the mouse collection. A Windows Precision Touchpad shall only report data by using one given collection at any time and shall only report from a different collection after the corresponding feature report has been received from the host that indicates the desired input mode.
> 
> The host may issue the input mode feature report to a Windows Precision Touchpad at any time after reading the report descriptor, including the time that data is potentially being reported through the active collection. In the event that a mode switch occurs while data is being reported, all contacts and button state should be reported as up and all reporting should cease by using that collection. Reporting by using the newly specified collection can occur after all contacts are physically up. The input mode shall not be persisted by a Windows Precision Touchpad across power cycles or host initiated resets (USB reset, HID I²C HIR); however the input mode may be persisted across any device initiated reset (for example, HID I²C DIR, etc.).

In English, it seems like MS is saying the following:

1. Start the trackpad in legacy mouse reporting mode.
2. Tell Windows that you are capable of both Mouse and Precision Touchpad modes.
3. Wait for Windows Switch between mouse/precision touchpad modes.
4. Whenever Windows switches, finish sending your old/buffered data using the old mode, and then start using the new mode.
5. The driver is NOT allowed to remember/persist the last-used input mode between computer restarts/sleep(?)/device driver restarts, except in some very specific circumstances.

So it seems like Microsoft expects every precision touchpad to also support legacy mouse mode and to follow the rules above.

imbushuo, 2020-09-11:
That is really optional in Windows 10. But yes I should add the legacy mouse fallback at some point.

---
Thanks,
Bingxing Wang

https://docs.microsoft.com/en-us/windows-hardware/design/component-guidelines/windows-precision-touchpad-required-hid-top-level-collections#input-mode-feature-report

Bananaman, 2020-09-11:
Ahh, understood. And yeah it works fine for me already in every scenario I've tried. Windows 10 does a great job emulating a mouse from the PTP input.

jsteweii, 2020-11-12:
Will there be an update for this ever? Thanks in advance:-).