imbushuo/mac-precision-touchpad #209: Need Help on Installation for MacBookPro15,2 (13" 2018 with Touchbar)

rbdej, 2019-07-30:
Hi, I've already installed it by following the instructions under the "Apple T2-based model Installation" at the main page. But i have an issue, I just cant seem to make it work. It says that operation completed successfully but I cant see any difference. Am I missing something, a step or setting maybe?

I have a Macbook Pro 2018 13" with Touchbar (MacBookPro15,2)

[image]

Some more details of what Macbook Pro I have:

Model Name:	MacBook Pro 13" 2018 with Touch Bar
Model Identifier:	MacBookPro15,2
Processor Name:	Intel Core i5
Processor Speed:	2.3 GHz
Memory:	8 GB 2133 MHz LPDDR3
Graphics:	Intel Iris Plus Graphics 655 1536 MB

Apple Internal Keyboard / Trackpad:
Product ID:	0x027b
Vendor ID:	0x05ac (Apple Inc.)
Version:	9.27
Serial Number:	FM7827301XGJ3VXAS+WRN
Speed:	Up to 480 Mb/sec
Manufacturer:	Apple Inc.
Location ID:	0x80500000 / 3
Current Available (mA):	500
Current Required (mA):	500
Extra Operating Current (mA):	0
Built-In:	Yes

phamminhtris, 2019-07-30:
Have you restarted your machine? Also, if nothing goes wrong, you can go to the trackpad setting and see if you find window precision features such as 3, 4 finger gestures.

imbushuo, 2019-07-30:
Check https://github.com/imbushuo/mac-precision-touchpad/releases/tag/1907-3540. 7a and 7b devices aren't added in the INF until today.

closed 2019-07-30 by imbushuo

rbdej, 2019-07-31:
> Have you restarted your machine? Also, if nothing goes wrong, you can go to the trackpad setting and see if you find window precision features such as 3, 4 finger gestures.

I have restarted it after installing and noticing nothing changes.

rbdej, 2019-07-31:
> Check https://github.com/imbushuo/mac-precision-touchpad/releases/tag/1907-3540. 7a and 7b devices aren't added in the INF until today.

Ohh okok, I'll try it out later this evening. Thanks again dev 👍 😄