imbushuo/mac-precision-touchpad #210: Driver Error on MacBookPro15,2 (Macbook Pro 2018 13")

rbdej, 2019-07-31:
Hey, Im currently typing this without a functioning trackpad on Bootcamp (although I actually know how to rollback the driver). I had just installed it using the latest driver released, but then Im getting this error:

I dont know what to do to install this properly so please send some help. Thanks again guys :D

[image]

imbushuo, 2019-07-31:
@rbdej Check https://github.com/imbushuo/mac-precision-touchpad/releases/tag/1907-3541

closed 2019-07-31 by imbushuo

rbdej, 2019-07-31:
> @rbdej Check https://github.com/imbushuo/mac-precision-touchpad/releases/tag/1907-3541

Ok ok, i'll test it again next week. Thanks for the full support 🙂

commit 1a88b189 (2019-07-31): Resolve #210