imbushuo/mac-precision-touchpad #212: Magic Trackpad 2 disconnects/freezes (+sensitivity issue)

eElor, 2019-08-06:
Thank you for your implementation, @imbushuo !
After installing it, my MT2 is working pretty good on Windows 10 x64. It also responds to custom gestures set over windows settings. However it seems to keep "disconnecting" - the cursor freezes and doesn't respond to the MT2 until I either move my mouse (still connected), click on the MT2 or wait for some time. Until now I couldn't recognize a pattern: it doesn't seem to happen after doing anything special (writing/selecting/...) it just freezes.
I also noticed that the sensitivity is a little low and it can't be regulated over windows settings.

krokofant, 2019-08-15:
I'm detecting this when two fingers have touched the trackpad. Moving one finger around and then touch somewhere with another finger.

eElor, 2019-08-23:
My observations so far: I have no idea how the implementation works, but it seems to be a problem with the "reactivity" and something like "refresh rate". Now that I am more comfortable using multitouch gestures, I noticed that the quicker I use gestures (less time among each of them), the more often the MT2 stops working and freezes. This gets worse the more fingers are involved. Further, the more I interact with the MT2 after freezing, the longer it takes to unfreeze. It even gets to the point where I have to unplug and plug it again for it to start working again. In some cases not even scrolling gestures work and sometimes the cursor completely freezes. 
Idea: introduce some kind of kind of refresh/restart if possible, since unplugging and plugging seem to revert all problems.

eElor, 2019-08-28:
Build 3561 seems to do the trick!
Thanks!

closed 2019-08-28 by eElor