imbushuo/mac-precision-touchpad #213: Not working on Macbook Pro14,3 (late-2017)

AlaricGilbert, 2019-08-07:
Not working on Macbook Pro14,3 (late-2017)
[image]
It says "One of the installers for this device cannot perform the installation at this time."

AlaricGilbert, 2019-08-07:
Hey, I've noticed that your driver targets to Windows 10 1803 environment just now and my system version is 1903, is the installation error caused by system version not compatible?

imbushuo, 2019-08-07:
@AlaricGilbert Drivers targeting 1803 works fine on 19H1. For MacBookPro14,3 you should follow the SPI installation guide as outlined in the README file.

AlaricGilbert, 2019-08-07:
@imbushuo I'm sure that I followed the SPI installation instructions in the README file but the error occurs :(

AlaricGilbert, 2019-08-07:
Update: I even tried to compile the driver myself in 1903 driver devkit and turned off sign-off verify, managed to get it installed but after installation nothing different to initial situation. QwQ.
I really want to be able to use it. Any solution?

imbushuo, 2019-08-09:
@AlaricGilbert Check the SetupAPI log. The error message doesn't make sense to me either.

AlaricGilbert, 2019-08-16:
Managed to get it working after reinstalled Windows.

closed 2019-08-16 by AlaricGilbert