imbushuo/mac-precision-touchpad #214: BSOD on install of build 3544

cocowalla, 2019-08-10:
I have a 2017 MacBook Pro (14,2), and just tried installing the x64 SPI driver (which I assume is correct for the integrated trackpad?), but got a BSOD after right-clicking on the .inf file and selecting "Install".

Error code was `DRIVER_IRQL_NOT_LESS_OR_EQUAL`

imbushuo, 2019-08-16:
`MT_Trackpad Filter Driver` is not my driver - have you installed Trackpad++?

closed 2019-08-16 by imbushuo

cocowalla, 2019-08-16:
@imbushuo ah 对不起, as the BSOD happened on install of this driver, I presumed it was this driver that caused the BSOD!

I actually tried installing this driver because I find Trackpad++ really glitchy - better than Bootcamp's crappy Windows drivers sure, but still _very_ annoying!

I'll try uninstalling Trackpad++ first, and *then* installing your driver.