imbushuo/mac-precision-touchpad #219: Magic Trackpad 2 via Lightning/USB on Win10 x64 doesn

Schroedingers-Cat, 2019-09-06:
**Describe the bug**
Using the "Apple Magic Trackpad 2" via a lightning connection on Windows 10 1903 x64, after installing the driver "AmtPtpDeviceUsbUm" or "AmtPtpDeviceUsbKm" as explained in the linked [reddit page](https://www.reddit.com/r/bootcamp/comments/c4alv2/how_to_install_mac_precision_touchpad_better_than/) leads to the error `Device cannot be started (Code 10)`. My Windows is non-english, so my translations might be slightly off the official ones.

**To Reproduce**
1. Enable testsigning via `bcdedit -set TESTSIGNING ON` as explained in the linked reddit post
2. Open device manager, go to "Input devices (HID)" and open the properties of the "HID-compatible Touchpad"
3. Go to driver tab and select "Update driver", click on "Search for drivers on this computer" and select "Choose a driver from a list" and there choose "Device" and there select one of the x64 drivers "AmtPtpDeviceUsbUm.inf" or "AmtPtpDeviceUsbKm.inf" and proceed
4. Confirm the following dialogue
5. The device's name changed to Apple USB Precision Touchpad Device (Kernel/User Mode), but you get an error Code 10 and 

**Expected behavior**
Installation should work and multi-touch gestures + right click should work

**Environment**
Apple Magic Trackpad 2, Lightning to USB connected, Windows 10 1903 x64

On a site note: even the reddit installation post is not accurate/up to date and getting this far took me an hour and was extremely confusing.

imbushuo, 2019-09-08:
- Only AmtPtpDeviceUsbUm is required. Test-signing is not required.
- Please capture a WPP/ETW trace and attach it here as outlined in the AzDevOps wiki.

Schroedingers-Cat, 2019-10-03:
I got it working using the AmtPtpDeviceUsbUm but installing the apple magic touchpad 2 drivers before. Is that expected?

imbushuo, 2019-10-04:
@Schroedingers-Cat You are only expected to use Magic Trackpad 2 with AmtPtpDeviceUsbUm but not the Km one. The Km USB driver only works with T2-based MacBooks and hence error 10 is expected.

closed 2019-10-04 by imbushuo

luvletter2333, 2020-07-19:
Magic Trackpad 2 with bluetooth connected to Windows 10 is not supported yet?

imbushuo, 2020-07-19:
Working on it (as I’ve told you)

luvletter2333, 2020-07-19:
oook. Happy to know that!

Thank you!