imbushuo/mac-precision-touchpad #225: SPI Build 3561 device not found error

markreg, 2019-09-22:
I'm unable to get the SPI Build 3561 driver working on a recent build of Win10. The touchpad disappears if I try to update the UMDF driver and says device not found before I rollback.

Currently using the USB UMDF driver, however it seems to have some significant issues with scroll sensitivity and lag.

imbushuo, 2019-09-24:
Need clarification on the driver you use, since USB and SPI use different implementation. UMDF driver has higher latency by design - this won't get mitigated until these drivers get merged in near future.

markreg, 2019-09-24:
Currently using AmtPtpDeviceUsbUm, x64 and I attempted to install AmtPtpDeviceSpiKm, x64.

First tried via "Have Disk" prompt from "Apple Precision Touchpad Device (User-mode)" in Device Manager, the install is successful but later it shows "device not found" code 10 and the cursor disappears.

imbushuo, 2019-09-24:
@markreg This is unsupported scenario. Until further notice, you should stay on AmtPtpDeviceUsbUm and never attempt to install AmtPtpDeviceSpiKm.

imbushuo, 2019-09-24:
But could you please provide the device ID?

closed 2019-09-24 by imbushuo

markreg, 2019-09-24:
Vendor ID: 05AC
Product ID: 0274
REV: 0624
MI: 02

markreg, 2019-09-24:
The primary issue seems to come when I _begin_ scrolling in either direction, there is a long delay and the cursor moves, rather than scrolls in most cases. Then the scroll motion is similar to what I experience on macOS.