imbushuo/mac-precision-touchpad #228: 无法在基于T2芯片的2019MBP上安装

leoleoasd, 2019-10-05:
For my convenience, I’m using Chinese. I can also provide a English issue, but my English is poor, so ask if you cant read Chinese.
**Pre-report checks**

- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
右键INF文件点击安装后， 没有任何有关确认开发者的弹框， 同时四肢划动没有任何反应

**To Reproduce**
我的安装过程： 1. 按照 [reddit](https://www.reddit.com/r/bootcamp/comments/c4alv2/how_to_install_mac_precision_touchpad_better_than/)上的教程，cmd中启用TESTSIGN
2. 尝试安装SPI设备的驱动（当时没有注意到readme里提到t2设备应该安装那个）
3. 提示签名问题，高级重启选择禁用强制驱动签名
4. 注意到readme中提到的应该安装USBKm驱动，我安装了/xxxUSBKm/x64/ReleaseSigned/中的cer证书文件
5. 安装同目录下的inf文件， 提示了“未知开发者”， 选择了确认， **没有弹框安装成功**
6. 按照reddit中的说明，四肢划动以测试驱动是否安装成功，没反应
7. 发现readme中要求安装xxxUSBKm/x64/ReleaseSigned/xxxUSBKm/中的inf文件，右键安装， **没有任何弹框**
8. 测试四肢划动仍然没有任何反应

注： secure boot已经关闭（安装windows的时候就关了，一直没打开）
**Expected behavior**
不知是否已经支持t2芯片的设备， 有提供应该安装哪个文件，但是下面仍然说不支持
**Traces**
不知如何获取， 如果告知如何获取可以提供

**Environment**
Add your device model, driver type (USB or SPI) and OS version here.
MAC：2019MBP 15寸 基于T2芯片
Windows ： windows10 LSTC 2019 **在windows to go中**

leoleoasd, 2019-10-05:
上面的四肢划动均为四指划动

leoleoasd, 2019-10-05:
Apple内置键盘/触控板：

  产品ID：	0x027c
  厂商ID：	0x05ac (Apple Inc.)
  版本：	9.27
  序列号：	FM79074012LHYYMAE+TDT
  速度：	最高可达480 Mb/秒
  生产企业：	Apple Inc.
  位置ID：	0x80500000 / 2
  可用电流(mA)：	500
  所需电流(mA)：	500
  额外的操作电流(mA)：	0
  内建：	是

imbushuo, 2019-10-05:
@leoleoasd It is not necessary to install the certificate manually. Windows requires the certificate to be placed in "Trusted Publisher", and thus manual installation may cause driver installation failure.

Open `certmgr.msc` with both your user account and computer account (Computer Management > Add snap-in > Certificate management > Computer Account), remove the certificate "Bingxing Wang" from your personal certificate storage, or move to the trusted publisher storage.

Then try installing it again and see if it helps.

leoleoasd, 2019-10-06:
看来就是证书的问题了,我卸载证书后仍然无法安装,安装spi的驱动就可以,正在重装系统试试(wtg,没重要数据)

readme中给出的reddit连接中给出的驱动安装视频里是安装了证书的, 个人认为有必要说明一下这一步不能做.

leoleoasd, 2019-10-06:
谢谢您,重装后安装成功了.

closed 2019-10-06 by leoleoasd

monisidd, 2022-10-07:
It is still giving Digital Signature issue.. Any leads?