imbushuo/mac-precision-touchpad #229: about MagicTrackpad 2 work via bluetooth

MistyMoonR, 2019-10-14:
First of all, thank you very much for your contribution.

As before, I used extra magic, but that doesn't work very well. So I was surprised to find this driver on GitHub. Roughly tested, the trackpad support is perfect on windows!!!

But there are minor problems, The MagicTrackpad 2 can only work with wired links, it can't work via Bluetooth. I make sure that the MagicTrackpad 2 connects to the computer's Bluetooth.  The second, but not important question is that MagicTrackpad 2 support for secondary pressure sensing on hardware, but it seem not be used on windows.

My PC is NUC not apple computer.

imbushuo, 2019-10-16:
This is on plan (CY20 Q1), but not currently implemented.

reznikartem, 2019-12-20:
Thank you! Send for you 5 coffees, and forget type message with it. 

Your work is awesome! Wait for MagicTrackpad 2 bluetooth support.

maffi-git, 2020-01-04:
I would also be happy to see this feature come to life. Was so unhappy to find out, it can't even handle right-clicks via bluetooth. 
Apple says its Microsofts fault. :-D 
I also have a programming background and also work on a hackintosh - If there is anything I can do to contribute...

asf5662004, 2020-03-21:
非常感谢您的贡献，同样期待蓝牙驱动的支持。

curamaj, 2020-04-01:
Hello @imbushuo ,

I am sure you are busy with development at the moment (whether related to this project or not).

I am new to this repository & have not used the driver that has currently been released in this repository. However, I do know that USB is working though as of this YouTube video: https://www.youtube.com/watch?v=vutQj7grdUA

In particular, I wanted to know about the progress of the Bluetooth driver to connect Magic Trackpad 2 to a Windows PC. I am keen to know what the progress is on this portion of the project. I currently have a Windows 10 PC on hand & a Magic Trackpad 2 that I wanted to use together, with BT in particular as opposed to USB. 

I read somewhere that you may not have access to all the HW you wish to develop your device drivers for, so if there is any way I may help you speed up development by using my machine & device for testing, or even letting me help with committing the source code needed to make this a reality please do let me know. I would be new to developing something like this though.

I have a programming background & use C/C++/Python extensively for both projects & work.

Don't hesitate to reach out to me if you need a hand. Really appreciate the work that you have done so far, and I am really looking forward to getting BT support enabled for Magic Trackpad 2 on Windows PCs.

If this is not the right place to have this conversation, please do let me know. Would be happy to talk about this elsewhere.

Regards,
Ajay

ericdhkim, 2020-04-04:
Is there a new timeline of when Bluetooth support is arriving?

imbushuo, 2020-04-05:
Actually it’s not yet changed; should be ready shortly within the month.

aalbul, 2020-04-18:
@imbushuo looking forward for it!
My wife is very happy that now she can use her mac touchpad at work as well.
Thank you for your effort!

Dani-O, 2020-04-19:
Thank you so much for your work! The trackpad works just perfectly via USB and I'm really looking forward to Bluetooth support. THANK YOU!

ericdhkim, 2020-05-06:
Any updates on this?

pravinkjana, 2020-05-08:
Thank you for creating this driver. Works well with USB. Looking forward to Bluetooth support.

Marak, 2020-05-09:
Any updates for bluetooth support?

imbushuo, 2020-05-10:
There's a P0 problem currently blocks it, will push once get that resolved.

---
Thanks,
Bingxing Wang

alexhaolin, 2020-05-13:
Go Go Bro! 奥利给！

danmogo, 2020-06-12:
Any updates on this? Really looking forward to it.

cktse, 2020-06-18:
Thanks so much!! Bootcamp drivers worked but control panel didn't show up so I lost the secondary click and scrolling gestures. 

Looking forward to the BT support!

jszroberto, 2020-07-07:
Same here!! Very good support for USB but I really miss the support for Bluetooth  :(

durafael, 2020-07-17:
The experience, is amazing when I use it. But, Bluettoth with will be better. Please, release the support for bluetooth. <3

claykohut, 2020-07-19:
Any word on the bluetooth progress? 

I'm trying to use the magic trackpad for my couch PC setup -- unable to plug it in to use over USB (and am super bummed about it)

karthi6662003, 2020-07-19:
eagerly waiting for bluetooth support

UlrikVH, 2020-07-30:
+1 for the Bluetooth support. Does someone know if it's currently in development? I would love to help where I can.

jpcercal, 2020-08-02:
I'm looking forward for the bluetooth support as well! USB support is amazingly well done, thank you guys. =)

y-rokutan, 2020-08-07:
Waiting for bluetooth support too. How can I support dev?

jszroberto, 2020-08-10:
+1 I could also support on the dev side.

claykohut, 2020-08-14:
@imbushuo any word here?

imbushuo, 2020-08-16:
@claykohut I pushed this back a few bit, but should come soon

cicerohellmann, 2020-08-20:
I'm waiting for it :)
This this is really great, I could also support in case of the lack of hands

Bananaman, 2020-09-10:
@imbushuo That's great to hear! :) My life will be improved when this driver is released for Bluetooth. I'm chronically ill (bedridden) and using Magic Trackpad 2 with a wire to my bed. When it's wireless, there will be less hassle for my legs to get trapped on cables. I already have a wireless keyboard. So I am really looking forward to a Bluetooth driver! :)

PS: I heard that Microsoft does not have any Bluetooth API for precision drivers. That's why the Corsair K83 wireless keyboard only has precision support when you use its USB dongle so that they can install the USB precision driver. So I'm surprised you're able to make Bluetooth precision work. Perhaps some magic with a virtual touchpad device driver that gets its commands from Bluetooth? Would love to hear how you manage to achieve this!

imbushuo, 2020-09-10:
There's a lot of things going on, but the work continues.

@Bananaman The claim is actually a hoax. Given a miniport driver, if you properly implemented required IOCTLs and PTP HID collections, you get a Precision Touchpad input. See SPI-based transport is not documented but we have SPI PTP anyway 🙂

---
Thanks,
Bingxing Wang

Bananaman, 2020-09-10:
@imbushuo Woah, that's very cool and surprising! You are a genius for figuring that out! :) My current USB cable is annoying and a bit short so I am really looking forward to the day that my bed setup is completely wireless. I have donated $10 from me (Johnny) on PayPal now, as thanks for your work towards this. You're going to be making a real improvement to my life when this great trackpad is wireless. Hopefully wireless will be achievable sometime this year. :)

Bananaman, 2020-10-02:
Hi again. Just curious. What does this closed Azure DevOps issue mean?

https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_search?text=Bluetooth*&type=workitem&lp=workitems-Project&filters=Projects%7BApple%20PTP%20Trackpad%7D&pageSize=25

imbushuo, 2020-10-02:
@Bananaman that's a mistake (cross referenced by other projects in the tenant.) Corrected.

Bananaman, 2020-10-02:
Hi again @imbushuo. I am very curious. What does this closed Azure DevOps issue mean? February 2020 "closing this issue since the work is complete"? Does that mean the driver just needs some polish and stability testing or something? Can we help in any way?

https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_search?text=Bluetooth*&type=workitem&lp=workitems-Project&filters=Projects%7BApple%20PTP%20Trackpad%7D&pageSize=25

imbushuo, 2020-10-02:
@Bananaman see previous comment.

Bananaman, 2020-10-03:
Ah alright. Thanks for clarifying. :) Thought it meant that the solution was almost complete. I am really excited for bluetooth support whenever it may happen (even if it takes some time), and understand that you'll be very busy after your graduation. If you ever need any public alpha testing, I'm here to help. Take care. :)

o000u, 2020-11-05:
> There's a lot of things going on, but the work continues. @Bananaman The claim is actually a hoax. Given a miniport driver, if you properly implemented required IOCTLs and PTP HID collections, you [...]

So glad to hear that！！
Really really need the BT !
Thank you!!!

ozskywalker, 2020-12-02:
Commenting here as I didn't see other doco on contribution.  I have a Win10 host w/Magic Touchpad2, connected via bluetooth and USB.  

Happy to provide testing & QA services, if that helps in any meaningful way, just ping back and let me know.

Zerov-ua, 2021-01-08:
Any progress?

SaschaJunghans, 2021-01-21:
I would be very happy about this working on bluetooth connected devices!

o000u, 2021-02-01:
Any progress?+1。。。
I hope my MT2 will still work when that day comes（TT）。。。

michaelzeng25, 2021-02-08:
Any update for Bluetooth support?

duckyb, 2021-02-08:
No need to ask for an update every few days, guys.

Just check on the [Azure DevOps - Feature 634](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/edit/634/) page, where you can see that the development of this feature is still active (in theory).

Latest updates from @imbushuo have been:

> should come soon 
\- August 2020

> the work continues 
\- October 2020

Clearly @imbushuo is still interested in porting this feature but has very limited time to dedicate to it and development is stalled / very slow.

I suggest you do not keep your hopes up for this feature coming any time soon, but do subscribe to this thread for future updates.

### 🙏🏻 Please don't spam with "_Any update_?" posts.

shaipetel, 2021-03-10:
This driver is great. Thanks!
I am pledging $20 donation once bluetooth is supported for window.

cicerohellmann, 2021-03-10:
Bounty

Bravatrue, 2021-04-30:
Just wanted to add a quick word of appreciation for still trying to implement this feature. I heard that creating Bluetooth drivers for Windows is apparently a massive headache, so I am quite grateful for your continued work on this.

qq84584, 2021-05-15:
> No need to ask for an update every few days, guys.
>
> Just check on the [Azure DevOps - Feature 634](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/edit/634/) page, where you can [...]

I'm not anxious about this update but just check the release website every few days. :)

imbushuo, 2021-05-29:
Well, after a while:

https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_git/Internal

This is not yet intended for your use (it's not functional yet.) But at least I can talk to Bluetooth in a sane way now.

imbushuo, 2021-05-31:
[image]

Soon (TM)

cicerohellmann, 2021-05-31:
Thanks man, I'm patiently observing your progress! More power to you!

closed 2021-06-01 by imbushuo

frichard35, 2021-06-02:
Thx you so much Ben for this feature,
you updated the README, but you don't put these latest drivers as a release. 
Is there any reason ? We can download them from the pipeline, they are there in the official version ?

imbushuo, 2021-06-02:
> Official version comes soon, expecting this Friday or Saturday.

FYI it's here: Release Release 3979 · imbushuo/mac-precision-touchpad (github.com)<https://github.com/imbushuo/mac-precision-touchpad/releases/tag/2105-3979>

---
Thanks,
Bingxing Wang

damianszocik, 2021-06-02:
@imbushuo, works like a charm on Windows 10 19042.508 🤩 Thank you!

seriva, 2021-06-02:
Just tried it, works great on my Thinkpad, however when connecting the Trackpad on my Macbook pro 16 inch with Windows on bootcamp it connects fine but the 3 and 4 finger gestures don't work. Works fine wired.

imbushuo, 2021-06-02:
@seriva seems like Boot Camp driver overrides it, you can double check if it's the case (something saying "Apple Multi-touch Pro"). Right click on that device, choose update driver, and choose HID Bluetooth Device in the list.

seriva, 2021-06-02:
@imbushuo Yeah that did the trick:D I just got this trackpad a week back in the hopes to use it across my Mac, Linux and Windows machines wireless and thanx to your hard working great! Enable to donate button so we can give back:)

shaipetel, 2021-06-02:
Nice! working perfectly!

Thank you!

huxmut, 2021-06-06:
what a champion 
is there a donate or paypal somewhere ?

adam-law, 2021-06-06:
I just wanted to add for those that installed this using Chocolatey and found that Windows set things to the Microsoft Bluetooth Device driver and it can't be changed using the Update Driver dialog in Device Manager.

1. Remove the device from the Devices list.
2. Check Device Manager if the trackpad is still listed under the Microsoft Bluetooth Enumerator. Uninstall it if it's there.
3. Turn off thetrackpad.
4. Download the latest drivers from the tagged release given above.
5. Extract to a folder that has an easy-to-use path.
6. Open a command prompt or PowerShell with admin privileges.
7. Input pnputil -a <path>\AmtPtpDevice.inf. Execute.
8. Turn on the trackpad.
9. Go to the Devices list and pair the trackpad. 

Everything should working (i.e. gestures) great.

ThiagoAnd, 2022-02-08:
Did you guys manage to get the bluetooth to work? I can connect with bluetooth, it shows magic trackpad 2 on bluetooth list, but there is no response from the trackpad when I try to use it.

MadMango, 2022-02-08:
> Did you guys manage to get the bluetooth to work? I can connect with bluetooth, it shows magic trackpad 2 on bluetooth list, but there is no response from the trackpad when I try to use it. Im using the last version of precision drivers

I've used it for a while, it works well. I'd make sure that you remove any other drivers first, there might be a conflict.

[1 more reply]