imbushuo/mac-precision-touchpad #230: Macbook Pro Mid 2012 not getting "Installation Succesful" and not working.

Chaos02, 2019-10-19:
Hello, I bought a used Mac Pro and from its Model Number I believe its a Mac Pro Mid 2012. I installed Windows 1903 with dualboot on it and all official Drivers through the Bootcamp installer.
Now I noticed that the touchpad is a Pain in the A** on it and I found these super drivers wich I tried out immediatly.
I am a Windows Prosumer and know and use nearly everything on Windows to make it work/better or customize or whatever but I cant get these to work.
I did the trial and error with many releases but the version that should work with mine (at least I believe so) didnt get a "Installation succesful" Window after I installed them, all others however did.
I always chose x64 and trialed my way to find out that it must be the USB-UM drivers for my macbook but it just wont work.
If you could assist me with this, I would be VEEEEERY thankful!
I have also tried the release that Azmat recommended in his better installation tutorial with the same Model but also didnt work.

Thanks in advance!

imbushuo, 2019-10-21:
@Chaos02 Could you provide your trackpad's VID/PID data?

Chaos02, 2019-10-21:
idk what you mean with VID/PID data but I will attach I found this in the driver protocol:
USB\VID_05AC&PID_0253&MI_01\8&1c8f4140&0&0001

imbushuo, 2019-10-21:
https://github.com/imbushuo/mac-precision-touchpad/blob/892bb92fab6eb99137030d7ee6cfd942b6e0671c/src/AmtPtpDeviceUniversalPkg/AmtPtpDevice.inf#L44

0253 is supported. Try the latest release.

Chaos02, 2019-10-22:
Thank you - it somehow worked, even tho the inf in the release folder didnt, the one in its subfolder "AmtPtpDeviceUniversalPkg" worked perfectly fine and I FINALLY have a good mouse - making backup of system instantaneously.
You could also maybe link to this here, as many people with bootcamp that may be using your  drivers. [Windows backup fails on Bootcamp](https://medium.com/@MadOverlord/cant-make-backups-on-windows-10-running-on-your-older-macintosh-fb654e96f04)

closed 2019-10-22 by imbushuo