imbushuo/mac-precision-touchpad #233: Touch pad totally not work

alexhaolin, 2019-10-22:
Touch pad has no response after Installed the AmtPtpDevice.inf
Before installation, have disabled the secure boot.

imbushuo, 2019-10-22:
Need your VID/PID combination so I can look into it.

alexhaolin, 2019-10-22:
Open the control panel and didn't find the touch pad. But I fond a unknown
device with a question mark on it. And the details of the hardware id is
below.
ACPI\VEN_INT&DEV_34BA
ACPI\INT34BA
*INT34BA
Hope this helps.

imbushuo, 2019-10-22:
`ACPI\INT34BA` is not the touch pad device. You should be able to find something under the HID section:

[image]

If there's no "Precision Touchpad" device and your trackpad is still functional, that means your model is not currently supported. In such circumstances, I need your "Apple SPI Trackpad" or "Apple Multi-Touch" device's ID so I can add support for it.

alexhaolin, 2019-10-22:
oops, found it. It's
USB\VID_05AC&PID_027B&REV_0927&MI_02
USB\VID_05AC&PID_027B&MI_02

imbushuo, 2019-10-22:
I see `027b` is supported by the INF:

https://github.com/imbushuo/mac-precision-touchpad/blob/892bb92fab6eb99137030d7ee6cfd942b6e0671c/src/AmtPtpDeviceUniversalPkg/AmtPtpDevice.inf#L34

Please verify if your trackpad is functional at this moment.

* If it is functional, you probably have to specify the proper driver using context menu (update driver). Apple drivers are WHQL signed and in some cases they have higher priority than my driver.
* If it is not functional, did the device report some problem (e.g. Error 10)? In such cases I probably need to collect some trace data.

alexhaolin, 2019-10-22:
Just installed again, restart, and update driver with your version, It
works. Thanks.

closed 2019-10-22 by imbushuo

ezamelczyk, 2019-11-08:
Mine is \SPI\VID_05ac&PID_0273&MI_02\ and it's not working. Each time I try to install the driver the trackpad stops working completely. What do?

ezamelczyk, 2019-11-08:
When I install the driver there is this error in the device manager:

```
Windows cannot load the device driver for this hardware. The driver may be corrupted or missing. (Code 39)

{Unable to Load Device Driver}
%hs device driver could not be loaded.
Error Status was 0x%x
```