imbushuo/mac-precision-touchpad #235: MacBook Air 2011 13 doesn't work.

1984dream, 2019-10-28:
Please add MacBook Air 2011 support, device id: USB\VID_05AC&PID_024D&MI_02

imbushuo, 2019-10-30:
@1984dream Would you mind providing physical dimensions (width and height) of the trackpad?

1984dream, 2019-10-30:
4.2x3.0 inch

imbushuo, 2019-12-01:
Correlated 12/1's remote debugging with this.

imbushuo, 2019-12-23:
@1984dream I pushed a update (not yet signed off the release but it is installation) for your device. Please try https://ligstd.visualstudio.com/7694e0d0-94e3-4fd2-b39a-ecd261e1ba2e/_build/results?buildId=3734 and see if it works.

1984dream, 2019-12-23:
> @1984dream I pushed a update (not yet signed off the release but it is installation) for your device. Please try https://ligstd.visualstudio.com/7694e0d0-94e3-4fd2-b39a-ecd261e1ba2e/_build/results?buildId=3734 and see if it works.

Works great! All gestures work, in the settings > devices > touchpad you can set every gesture, sensitivity, just like in macOS, although when scrolling on macOS it's just a bit more sensitive but not too different, overall I'd say it works perfect, thanks.

anohren, 2020-01-01:
> overall I'd say it works perfect

Do you experience the same issue as me where a right click fails to register about 50% of the time if you move the cursor with one finger, then stop moving but with the finger still on the touchpad add a second finger and press down in order to make a right click with the physical switch?

To me they register as single finger clicks very often, resulting in e.g. following a link instead of displaying its context menu.

1984dream, 2020-01-04:
Yeah, might be, but I have never used it in that way, isn't just tapping with two fingers way faster?

anohren, 2020-01-04:
> isn't just tapping with two fingers way faster?

Regardless of whether it is, faster does not equal better in terms of usability which is why I don’t think that detail matters. 

For me personally, no. Two finger tapping requires lifting the initial finger first — adding a second finger does not. 

I just wanted to gauge the extent of the bug. Maybe it just occurs on some touchpads.  

I guess I’ll add it as an issue.

closed 2020-03-02 by imbushuo