imbushuo/mac-precision-touchpad #236: Uninstall drivers

ciceroneves, 2019-10-28:
**Pre-report checks**
- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Is your feature request related to a problem? Please describe.**
I would like to have the option to remove/uninstall all the files installed by the .ini file

**Describe the solution you'd like**
Some clear instructions on how to remove the files in the README file would be enough

**Describe alternatives you've considered**
I considered looking into what the ini file does. But I am not an expert so it would be nice to have some instructions

MagicUtilities, 2019-10-28:
You can use Windows bult-in **pnputil** from the command line with administrator rights. 

- **pnputil /enum-drivers** lists all installed drivers. 
You need to find the correct one and remember the belonging **oem???.inf** 
- **pnputil /delete-driver oem???.inf /force** will delete the driver from the Windows driver store.

imbushuo, 2019-10-30:
@ciceroneves Apart from the mention used above, you can right click the device in Device Manager, select "Uninstall Driver" and also check remove driver package. Then re-scan devices and the trackpad will be reverted to the stock configuration / driver.

closed 2019-10-30 by imbushuo