imbushuo/mac-precision-touchpad #237: Magic Trackpad 2 too sensitive (1mm from surface)

tyger, 2019-11-01:
**Pre-report checks**

- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
Trackpad reacts on my finger that is in 1-2mm distance from trackpad surface. So basically I can move cursor not touching trackpad at all. Feels like it's too much sensitive (no such behaviour with this trackpad when connected to MacBook)

**To Reproduce**
Steps to reproduce the behavior:
1. Connect device to Windows laptop and intall driver
2. Try to move finger within 1-2mm from surface.

**Expected behavior**
Trackpad should react on the finger only and when it is touching surface of trackpad.

**Traces**
If applicable, add WPP traces to help explain your problem.
????

**Environment**
Apple Magic Trackpad 2 (Model A1535) 
connected to USB to Lenovo X1 Extreme
Windows 10 v1903

driver type is USB type (User-mode)

imbushuo, 2019-11-07:
Currently the threshold is hard-coded, a tool for the adjustment is planned.

Also relate to #239.

tyger, 2019-11-08:
@imbushuo  What is your idea about making it adjustable?
Will it be easier to just put something into Windows Registry? That should save some time creating tools.

imbushuo, 2019-11-11:
Have thought of that. Probably need to do some application work.

larryqiann, 2019-11-27:
It looks like there is a sensitivity setting under Windows settings app. Could this perhaps be used to set these settings?

reznikartem, 2020-01-09:
+1 for this issue. Mayby, because of that i often get cursor "shake"

Here is short video with demostration: https://puu.sh/EY0jZ/2f1166eb5a.mp4

tyger, 2020-01-09:
@larryqiann 
> It looks like there is a sensitivity setting under Windows settings app. Could this perhaps be used to set these settings?

I don't think it's the same as settings on Windows level influence all touchpads in the system (and I'm using laptop, so have an extra one that is build in)
It must be something on the device level.

tyger, 2020-01-09:
@reznikartem 
> +1 for this issue. Mayby, because of that i often get cursor "shake"
> 
> Here is short video with demostration: https://puu.sh/EY0jZ/2f1166eb5a.mp4

Yes, that looks like something that I experience some time as well.

evi0s, 2020-04-24:
Same issue. Any idea about changing the threshold? Maybe we can modify the source code and compile it manually before the tools release.

taterujp, 2020-05-01:
I also have the exactly same issue. Changing the sensitivity value in Windows setting did not affect this issue at all. Any other solutions?

taterujp, 2020-05-07:
I also tried the Magic Utilities then the sensitivity was ok with it. However some important behavior are not supported (dragging by tap-hold-move, dragging by 1stfingerclick-2ndfingertapholdmove, horizontal scroll on Excel and other apps, etc) and it isn’t enough for me to use.
Again, is there any way to change this sensitivity?

omegamalkior, 2020-07-12:
> I also tried the Magic Utilities then the sensitivity was ok with it. However some important behavior are not supported (dragging by tap-hold-move, dragging by 1stfingerclick-2ndfingertapholdmove, [...]

It seems to only happen with this specific precision driver. If you disconnect from it and use it with the native bluetooth or native USB support without the driver this issue does not happen. With touchpad++ driver and with the other paid driver it does not happen. Only with the Precision driver does it specifically happen. Makes it a bit unstable overall for daily use. Hopefully a fix isn't that far away.

joeholley, 2020-08-14:
Yes, as Evi0s mentioned above, since this has gone a long time without a fix, maybe you could let us know which line has the value so we could adjust/recompile manually for the time being?

michaelgosling, 2020-10-29:
Would love a fix for this as well. Coming up on a year with no fix now. The driver is fantastic in every other way but this hyper sensitivity can be frustrating.

funky-jojo, 2021-06-02:
@imbushuo Any updates or timeline for providing a way to adjust the threshold?  This is still a major issue, and prevents me from using it or promoting this driver to others.  For me it's overly sensitive, and the cursor moves easily without even physically touching the surface.

duckyb, 2021-09-07:
> Yes, as Evi0s mentioned above, since this has gone a long time without a fix, maybe you could let us know which line has the value so we could adjust/recompile manually for the time being?

https://github.com/imbushuo/mac-precision-touchpad/blob/734b084b365b6168234049a8a61e569ae5ab19ca/src/AmtPtpHidFilter/include/Metadata/MagicTrackpad2.h#L8

https://github.com/imbushuo/mac-precision-touchpad/blob/734b084b365b6168234049a8a61e569ae5ab19ca/src/AmtPtpHidFilter/include/Metadata/MagicTrackpad2.h#L49

It's probably these two lines that affect it. Good luck if you try to compile it!

imbushuo, 2021-09-07:
@duckyb Not really that line, there was two hardcoded threshold things in HID `Input.c`:

Check how tip switch bit and confidence bit is reported, and currently it's hardcoded. On high level I just need to make these two values configurable.

duckyb, 2021-09-07:
@imbushuo Thank you for the correction, so it's this one?

https://github.com/imbushuo/mac-precision-touchpad/blob/dd364d6c432bb2b1119c966046044e4068802ed4/src/AmtPtpHidFilter/Input.c#L214-L217

@funky-jojo I'd love to help but I don't have the required skills to do this, unfortunately. I've never programmed in C or made drivers before.

imbushuo, 2021-09-07:
@duckyb Yes.

MagicUtilities, 2021-09-08:
As a general hint (speaking for the Magic Trackpads only but assuming it is valid to any Apple Trackpad), a touch also has a "State" which precisely describes if a touch is nearing, hovering, touching, leaving,... 

Note: Testing Apples' recent PTP driver, it ignores that state as well (which makes you wondering if Apple deliberately cripples their drivers to make Windows users feel worse than macOS users). Hope this helps.

tritegeist, 2021-09-26:
To anyone interested - I just submitted a PR that I believe would fix this issue:

https://github.com/imbushuo/mac-precision-touchpad/pull/436

I tested it out and it works perfectly, at least on my Magic Trackpad 2. Building it is a bit of a pain so hopefully we can get an official release here soon. 🤞

henusys, 2022-11-10:
Is there someone who could repair this function? This driver is very useful, but it's not perfect.

orn1983, 2025-04-13:
Bump :) This is my only gripe.

mmikeww, 2025-04-13:
Native Windows option coming soon for us:

https://www.theverge.com/2025/1/5/24334270/hyperspace-trackpad-pro-apple-magic-trackpad-ces-2025

funky-jojo, 2025-04-13:
I’ll believe that when I see it. Windows touchpads are notoriously crappy.