imbushuo/mac-precision-touchpad #239: Touchpad Not sensitive enough (Late 2018 MBP)

p4xcheats, 2019-11-05:
**Pre-report checks**

- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
I can lightly glide my finger over the touchpad and the cursor doesnt move.

**To Reproduce**
Steps to reproduce the behavior:
Lightly run your finger over the touchpad in macos, then do it in windows. You will notice that macos is more sensitive.

**Expected behavior**
A clear and concise description of what you expected to happen.
It should be as sensitive as macos

**Traces**
If applicable, add WPP traces to help explain your problem.

**Environment**
Add your device model, driver type (USB or SPI) and OS version here.
Late 2018 Macbook Pro. OSX Catalina and Windows 10

imbushuo, 2019-11-07:
Currently the threshold is hard-coded, a tool for the adjustment is planned.

manolz1, 2020-01-16:
Having this issue as well on my MBP 2018 using the latest version of the drivers. I think its because of the surface area of the finger. When I use my entire fingertip it seems t to work fine, but when using just the tip of my fingertip it doesn't register at all. Is it possible to configure the surface area of touchpad recognition (for it to work with less surface area from the finger)?

Also have a 1cm deadspot from the top, left and right sides of the trackpad. Bottom works fine.

theprovidr, 2020-01-30:
I have the exact same issue on my Macbook Air 2018 model. No response if you slightly run you finger over the trackpad. However, it works well if you put a little bit pressure. Like @manolz1 said this seems to be an issue of surface area of the finger. Less surface area = No response. If you could fix it, would be grateful.