imbushuo/mac-precision-touchpad #240: Caution:Uninstall Trackpad++ driver before installing this

vishal-ph, 2019-11-11:
**Pre-report checks**

- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Traces**
If applicable, add WPP traces to help explain your problem.

**Environment**
Add your device model, driver type (USB or SPI) and OS version here.

vishal-ph, 2019-11-11:
If you have tried trackpad++ driver in the past on your PC and now want to switch to this driver. Information below is important.

Before, you install this driver it is crucial you uninstall trackpad++ app from program files and also the drivers from C:/Windows/System32/DriverStore. If you do not do this, you will lose your mouse pointer right after installation. To prevent this, make sure you have removed both the app and the drivers from trackpad++

imbushuo, 2019-11-11:
See also: https://github.com/imbushuo/mac-precision-touchpad/issues/214

imbushuo, 2019-11-11:
This is a Trackpad++ issue, so what I can do is limited.

Pinned for further references.

closed 2019-11-11 by imbushuo

shudabi, 2019-11-30:
Sorry for my carelessness. I just did not uninstall trackpad++ before the installation and it turns out that my touchpad cannot work. Is there anyway to repair this？

imbushuo, 2019-11-30:
@shudabi Use an alternate mouse to do a complete Trackpad++ driver removal. Then go to device manager, remove the trackpad device (also check remove driver package), rescan changes and reinstall the driver.

shudabi, 2019-11-30:
It works!!! Thanks for your help.

franciscobouza, 2019-12-18:
I made the mistake of installing TrackPad++ before hearing about this wonderful driver... I installed it while I had TrackPad++, then uninstalled TrackPad++ and tried installing this driver again... I don't see this working... Should I uninstall this driver and install it again?

christinem, 2020-07-26:
> @shudabi Use an alternate mouse to do a complete Trackpad++ driver removal. Then go to device manager, remove the trackpad device (also check remove driver package), rescan changes and reinstall the driver.

This worked perfectly for me, Trackpad++ had an error when installing so I decided to just switch to this, and experienced the problem of my trackpad not working anymore. Following these instructions using an external mouse worked perfectly. Thanks!

NightMachinary, 2020-11-03:
You need to add this in bold to the readme. It's very serious.

BTW, you do not need a mouse to uninstall the trackpad++ driver. It's possible with just the keyboard (play with the keys TAB, ALT, SPACE, ENTER, WINDOW to get the hang of it.)

MateoKngs, 2020-11-05:
> @shudabi Use an alternate mouse to do a complete Trackpad++ driver removal. Then go to device manager, remove the trackpad device (also check remove driver package), rescan changes and reinstall the driver.

I have uninstalled the Trackpad++ app. But not able to remove the driver. I'm logged in as the admin user, and also I have tried removing from CMD, but still says access denied, or SYSTEM permission needed, or logged on user does not have ownership privileges. What the hell is this issue!

MateoKngs, 2020-11-05:
> @shudabi Use an alternate mouse to do a complete Trackpad++ driver removal. Then go to device manager, remove the trackpad device (also check remove driver package), rescan changes and reinstall the driver.

I don't know how to thank you. First I installed your driver and my MacBook's screen went blue. Then I read that I must first delete Trackpad++'s driver. After hours of struggling to remove Trackpad++ driver(because of permissions' issues), I removed that and installed your driver. Then I lost my mouse pointer as you said(😅) and then I went to Device Manager and updated the touchpad driver and now everything's fine.

shad-1, 2021-01-26:
If you read this thread wondering if it will work, here's my experience: 

- Went to settings, uninstalled both programs with "Trackpad++" in the name
- When I went to the above file location to double check, there was no Trackpad++ driver 
- Installed via chocolatey, ran the script

And immediately, it began working. This is much better!

kychris, 2021-02-26:
@shad-1 yep! that works, thanks!

@imbushuo thank you so much for this incredible driver! so amazing!

hfake35, 2021-07-02:
The Trackpad++ Driver in driverstore was somehow protected by Windows and impossible to delete. The free software driverstoreexplorer (also hosted at github) solved the problem for me. 

BTW Thanks for this nice software. I appreciate it.

mehdirazavi, 2022-06-03:
completely mac trackpad experience on windows . thanks you a lot