imbushuo/mac-precision-touchpad #241: 'Left click' when index finger tap while middle finger still touches touchpad(not right click)

maximser, 2019-11-18:
I am using this driver on my MacBook Pro retina (15" late 2013). 
In macOS beahavior: moving cursor with middle finger, and when you want to 'left click' something, you just tap with index finger (middle finger still touch the touchpad).

Is there a way to do the same in windows with your driver?

reznikartem, 2019-12-24:
I'm faced same problem with magic trackpad 2 - right click with one  index finger, when big finger touches trackpad at bottom. Or, when big finger touches bottom, and i'm try to do right click with index + middle fingers, i'm make three fingers click :( same problems with unexpected gestures (three or four fingers swipes left/right/top/bottom)

anohren, 2020-01-10:
This is functionality I didn't know existed. Now I have to try it out.

DrJedikiah, 2020-01-10:
It works but for a second left click you have to release your finger.

alexporrello, 2023-01-15:
Is this a duplicate of https://github.com/imbushuo/mac-precision-touchpad/issues/221?