imbushuo/mac-precision-touchpad #242: Driver does not seem to be working with Macbook Pro 16 2019

YijieDeng, 2019-11-21:
**Describe the bug**
Followed readme.md to install on Macbook Pro 16 2019, no Secure Boot, certificate seems to be no problem, but the system does not recognize it as precision enabled device. Not sure how to check if driver is correctly installed.

**To Reproduce**
1. Do a bootcamp install, config nothing afterward.
2. Follow readme "Converged Installation Guide" and stop before "For developers". 

**Expected behavior**
Windows will not recognize touchpad as precision enabled as before installation. 

**Traces**
Not sure how to trace.

**Environment**
Macbook Pro 16 2019, built in trackpad (SPI I guess?), Windows 10 OS number will be posted here soon, as either the current bootcamp or "installing certicificate" stops Windows menu search bar functions. (Still questionable about the cause.)

_Note_
As BootCamp support was just released last day, I am not expecting a quick fix. But please take a look in this! Thank you!

imbushuo, 2019-11-22:
@YijieDeng All T2 trackpads use the UsbKm driver instead of the SPI one. So support will be added shortly.

carrottopadc, 2019-11-22:
Found this threat for the 16 inch macbook that i just bought since i used previous drivers on my 15 inch. Looking forward to the update that adds support for the 16 inch!

imbushuo, 2019-12-01:
@carrottopadc @YijieDeng Can you provide the VID/PID information for 16-inch MacBook Pro?

carrottopadc, 2019-12-01:
> @carrottopadc @YijieDeng Can you provide the VID/PID information for 16-inch MacBook Pro?

Can you point me in the direction on how to give you that information? Thanks!

imbushuo, 2019-12-01:
@carrottopadc 

[image]

carrottopadc, 2019-12-01:
[image]

carrottopadc, 2019-12-01:
@imbushuo Hope that helps!

Just wondering, why do you need it/what is it for?

imbushuo, 2019-12-01:
@carrottopadc I need that to match it in driver manifest. All T2 devices should have a generic driver, so you should be able to use it very soon.

carrottopadc, 2019-12-01:
@imbushuo oh awesome! Thank you so much for looking into making drivers for the 16inch mbp, I'm sure everyone really appreciates it

gnawh, 2019-12-05:
Can't wait to try it! Thanks for your work.

cubriq, 2019-12-11:
I am also eager to test this on my new MBP 16" Win 10 pro! 
Let me know if I can be of any assitance...

imbushuo, 2019-12-13:
PID 0340 for 16inch should be supported with https://ligstd.visualstudio.com/7694e0d0-94e3-4fd2-b39a-ecd261e1ba2e/_build/results?buildId=3724, please try and see if it works.

Make sure you have an alternate USB mouse so in case if things screw up, you can remove the driver in device manager by:

* Expand HID device category
* Find the Precision Touchpad Device and right click remove device
* Also checks the remove driver option

You may experience a "dead zone" issue due to uncertainties in trackpad dimension (physical and logical). If that happens, please let me know.

cubriq, 2019-12-13:
Thanks! Cool I will try it tonight and let you know the result. I will get back to you soon.

Is it this one? \Drivers\AmtPtpDeviceUniversalPkg\x64\ReleaseSigned\AmtPtpDeviceUniversalPkg\AmtPtpDevice.inf

I will follow this guide: 

Converged Installation Guide

1. Disable Secure Boot if your MacBook comes with Apple T2 Security chip: Boot with Command + R, select Startup Security Tools and disable Secure Boot (No Boot Security)
2. Go to the release tab in Github and download the newest version
3. Open the zip and navigate to x64\ReleaseSigned
4. Go to AmtPtpDeviceUniversalPkg directory
5. Right click AmtPtpDevice.inf and install it
6. Confirm driver publisher

Happy Holiday to you too!

imbushuo, 2019-12-13:
Yes the procedure and file looks correct.

Thanks,
Bingxing Wang

Get Outlook<https://aka.ms/qtex0l> for iOS

nickaroot, 2019-12-13:
Hey there
Tested for dead zone
Just for 1 cm on left, top and right corners there is no initial mouse move
And no more moving for continuous swipe from center in parallel direction (top - vertical, left, right - horizontal)

cubriq, 2019-12-13:
Hi! It installed successfully on my MBP 16" PID 0340! Scrolling and gestures generally feels fantastic!

However I also found the dead regions, see picture here: 
https://www.figma.com/file/WLpzBqYfTcyGICCdizzmij/Touchpad

Once again thanks a lot for developing this! 

BR / Richard

cubriq, 2019-12-13:
...I can clarify:

- Clicking in the dead regions is working, 
- Moving the pointer is not working (single finger)
- Scrolling with two fingers is working everywhere except in the top red area of the picture

cubriq, 2019-12-13:
And one more observation. After browsing websites for a few minutes it seems as the scrolling gets less smooth. Acceleration not always happening. For example Apple webiste scrools very smooth at start but after a while the smooth acceleration stops working. Dont know if this is something that other users have experienced?

imbushuo, 2019-12-16:
A few websites behave weird and that's not the trackpad's issue. I will push a fix for the dead region shortly.

cubriq, 2019-12-16:
Thanks a lot! No stress and good luck with your exam! I will make sure to sponsor your project by PayPal...

carrottopadc, 2019-12-16:
Hello, just wondering if it is good to download and use https://ligstd.visualstudio.com/7694e0d0-94e3-4fd2-b39a-ecd261e1ba2e/_build/results?buildId=3724 on my 16inch mbp? Any major known issues? Or is it just testing the waters and such?
Cheers!

imbushuo, 2019-12-16:
@carrottopadc The most significant problem is dead zone (see the thread). It is production signed and you can actually use it.

kylekoren, 2019-12-18:
@imbushuo Just wanted to say thanks for all your hard work on this. Worked like a charm on my 16" MBP. Looking forward to final release!

cubriq, 2019-12-18:
I have tested the driver for a few days now and general impression is very good. Gestures works fine (Much better than Trackpad++). The dead regions dont bother me.  

My only concern is still scrolling. The bootcamp default driver is terrible for scrolling so it is a BIG improvement but it seems have some issues with the acceleration. 

After reboot the scrolling works perfect but after a while a gets less smooth (Both in windows and in Google Chrome web browser). It happens that the acceleration stops working for a few "pushes" and then comes back again after a while.  Could it be a memory leak? 

I have checked task manager and both CPU and memory looks fine. I also disabled Windows defender temporary but no difference. I have a new clean installation of Win 10 Pro. Have anyone else experience scroll acceleration problems on T2 device?

bermudalocket, 2019-12-29:
I'm experiencing the above as well. Unfortunately I don't have much to add: I'm not familiar with the details of Windows' driver/kernel interactions, and I have the exact same setup as @cubriq, being a MBP 16,1 on Windows 10 Pro freshly installed via Boot Camp. The only new information I can report is that it was only happening to me when two-finger swiping downwards. I'll edit this comment if that changes.

I echo the others' thanks. This software makes using Windows much more bearable.

hwangjason, 2020-01-07:
Deadzones are killing me. It's about 2 cm around. 
Also, anyone else's scrolling too slow? Is there any way to speed this up that I'm not seeing?
Thank you for your work!

imbushuo, 2020-02-26:
Initial deadzone fix pushed with https://github.com/imbushuo/mac-precision-touchpad/commits/2002-input-trace.

imbushuo, 2020-02-26:
Test this version: [AmtPtpDeviceAmd64.zip](https://github.com/imbushuo/mac-precision-touchpad/files/4257413/AmtPtpDeviceAmd64.zip)

For testing purpose, it is not MS signed (but signed with my code signing). So you still need to turn off Secure Boot for testing, but `TestSigning` is not necessary.

cubriq, 2020-02-26:
Thanks! Good job, dead zones are gone now and gesures works everywhere on the touchpad. Scrolling is still a bit "jumpy" if trying to scroll fast up and down. But in general it works fine. 10 times better than then horrible original bootcamp-driver without "precision". Thanks again

imbushuo, 2020-02-27:
I've pushed the Microsoft signed packages to release section.