imbushuo/mac-precision-touchpad #244: Right button(two fingers) drag

Firstgoer, 2019-12-08:
**Pre-report checks**
[Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for duplicates?
- [x] checked

**Describe the bug**
* Dragging with 2 fingers is not working (emulated right button). It is using left(single finger) dragging instead. 
* Dragging is not working with enabled `Tap with two fingers to right-click`. Coursor is not moving.
* Dragging is working for `Press the lower right corner of the touchpad to right-click`. But it is working for single finger also.

P.S. This [issue]( https://github.com/imbushuo/mac-precision-touchpad/issues/206) also have same bug, but it is more likely connected to old hid devices, so I created separate one.

**To Reproduce**
Steps to reproduce the behavior:
Try to drag any file or folder on window. Right-button frag should be completed with opened dropdown menu. Also, testing can be done in any application, using right button drag (World of Warcraft turn behaviour, for example)

**Expected behavior**
Dragging is working as it was done with right clicked mouse button

**Environment**
Device model: USB\VID_05AC&PID_0274&REV_0624&MI_02
(mid-2015 macbook pro), driver type: SPI, OS version: Windows 10 x64.

imbushuo, 2019-12-13:
I think this is a Windows behavior, I will try to reproduce it on Surface Pro later.

Firstgoer, 2019-12-14:
Default bootcamp drivers works same as on my ThinkPad. When user performs right click, dragging is disabled

imbushuo, 2019-12-14:
Wait - how do you using Boot Camp driver on a ThinkPad?

Firstgoer, 2019-12-15:
Default Boot Camp drivers(on MacBook) works same as native touchpad drivers on ThinkPad.

imbushuo, 2019-12-15:
Need to confirm your ThinkPad trackpad - I am aware that most of them are not PTP compliant.

Thanks,
Bingxing Wang

Get Outlook<https://aka.ms/qtex0l> for iOS

Firstgoer, 2019-12-15:
I think it is not the case. The message here - other drivers don't use right button drag, so it is not a bug, but enchantment :-(

imbushuo, 2019-12-15:
Since the driver is PTP compliant, gestures’ behaviors are determined by Windows. Given the current driver architecture it is difficult to implement such enhancement. I will continue look into it, while you can file a feature suggestion in Windows feedback app.

Thanks,
Bingxing Wang

Get Outlook<https://aka.ms/qtex0l> for iOS

closed 2019-12-18 by Firstgoer