imbushuo/mac-precision-touchpad #245: How do I uninstall the precision touchpad? (Touchpad dead)

BeepBoopMeow, 2019-12-08:
I am using Macbook Pro 2018 15inch.
I accidentally skipped procedure 0 (Disable Secure Boot if your MacBook comes with Apple T2 Security chip: Boot with Command + R, select Startup Security Tools and disable Secure Boot (No Boot Security))

I installed the newest version of precision touchpad, and after I reinstalled bootcamp drivers (I had sound driver problem. Reinstalled to fix it).
After restart, my touchpad doesn't respond anymore.
I just want to tryit over again. But don't know how to delete the driver.

So How can I delete the driver?

imbushuo, 2019-12-08:
- Open Device Manager use an alternate mouse
- Find either "Apple Multi-touch" or "Apple Precision Touchpad Device" under HID device category
- Right click "remove device", also check "remove driver"
- Rescan device changes using the menu

closed 2019-12-08 by imbushuo

b-hayes, 2020-04-02:
This should be in the readme.