imbushuo/mac-precision-touchpad #248: The third party inf does not contain digital signature information

mehdien, 2019-12-18:
hi guys, I am encountering a problem installing this driver, 
when i right click and install the amtptpdevice.inf on 

C:\Users\Mehdi\Downloads\mac-precision-touchpad-master\mac-precision-touchpad-master\src\AmtPtpDeviceUniversalPkg

i get this error message
The third party inf does not contain digital signature information

am i doing something wrong ?

MATERIAL
Macbook pro 2016 15 
windows 10 1863

[image]

mehdien, 2019-12-18:
Can someone please do a How to install video with all the steps starting from the download and post it somewhere on YouTube ?

imbushuo, 2019-12-23:
You are downloading the source code. Please proceed to release section.

closed 2019-12-23 by imbushuo