imbushuo/mac-precision-touchpad #252: Some fatal installation error in build 3665, The system cannot find the file specified

edimetia3d, 2020-01-05:
**Pre-report checks**

- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
I am installing driver for a fresh new windows 10 LTSC with macbook pro 13 2015 mid.

When I right-click the `.inf` to install, it says "系统找不到指定文件" (The system cannot find the file specified).

I checked the setup log, it turns out that the installation procedure tried to copy a file "Drivers\AmtPtpDeviceUniversalPkg\x64\ReleaseSigned\AmtPtpDeviceUsbUm.dll", and apparently, it's not there.

I don't know if there is other problem, so I just used the build 3656 instead.

edimetia3d, 2020-01-05:
Also, I would recommend to update the installation guide.
1.  I have to disable driver signature verification to install build 3656
2. The driver is not actived automatically after reboot, I need to selected it manually in device manager.

imbushuo, 2020-01-05:
If you have to force install the driver w/o signature verification - this means you are choosing the wrong INF file (there are multiple copies).

[image]

This one is what you want. I will improve the CI package to eliminate the confusion later.

closed 2020-01-05 by imbushuo