imbushuo/mac-precision-touchpad #253: can't find .inf file

airpods1, 2020-01-06:
Hello,
In any of the unzipped driver files I can't find AmtPtpDeviceUsbUm.inf. I don't know what driver to install

dunestorm333, 2020-01-06:
Assuming you're on 64-bit version of Windows, you need to right click and install **AmtPtpDevice.inf** from the extrated path **Drivers\AmtPtpDeviceUniversalPkg\x64\ReleaseSigned\AmtPtpDeviceUniversalPkg**.

airpods1, 2020-01-06:
I don't know if there's something I'm not seeing. 

[image]

[image]

dunestorm333, 2020-01-07:
> I don't know if there's something I'm not seeing.
> 
> [image]
> 
> [image]

That's the one the second entry down, you need to right click and install it.

airpods1, 2020-01-07:
I don't know what happened but now it's showing the correct .inf files. However, now when I try to right click I don't the option to install?

[image]

dunestorm333, 2020-01-07:
> I don't know what happened but now it's showing the correct .inf files. However, now when I try to right click I don't the option to install?
> 
> [image]

That's not right, there should be an option to install visible on INF files when right clicking them. Not sure where to go from here other than to try and find the correct trackpad device in device manager and install it from there.

airpods1, 2020-01-07:
I haven't activated Windows (I'm current running the free version) so my limited access to a lot of features might be inhibiting things? Not sure.

anuscleaner, 2020-01-07:
think u might be clicking the file while it's still in the zip file. try extracting it somewhere then right click, you should see the install option

airpods1, 2020-01-07:
Yes, it works now! I wasn't extracting it but instead just opening the zip file. 
Thanks so much dunestorm33 and anuscleaner!

closed 2020-01-16 by imbushuo