imbushuo/mac-precision-touchpad #254: Support for Macbook Pro Mid-2010 (MacBookPro7,1)??

anuscleaner, 2020-01-07:
Not sure if this is the right place to ask this but...

Asking for support on this specific macbook pro, have tried many of the drivers and none have worked so far. Does anyone happen to have the same model?

imbushuo, 2020-01-16:
I don't think I implemented support for MacBookPro7,1. What's your device VID/PID?

anuscleaner, 2020-01-17:
ayyy thanks for replying

[image]

here ya go :)

omscarr, 2020-02-09:
Not sure if I should open another ticket for this, but I also have a mid-2010 MacBook Pro, but mine is a 15" (MacBookPro6,2)

imbushuo, 2020-02-09:
0x0236 is defined in the source but have not yet exposed via the INF file. I think I can add support for it shortly. May I know the physical dimension of the trackpad?

anuscleaner, 2020-02-11:
my trackpad's 10.5 x 7.6 cm

imbushuo, 2020-02-16:
Thanks for the info

imbushuo, 2020-03-02:
Please try https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_build/results?buildId=3806&view=results and see if that works. If so, I am going to close this issue and push a new release.

MrTeoKuzma, 2020-03-31:
Hey there!
Could you also add support for the White Macbook 2010  (13'' not PRO), the size is the same and all but my product ID is: 0x0237
Thanks!

imbushuo, 2020-03-31:
@MrTeoKuzma It's in the build mentioned above.

anuscleaner, 2020-05-17:
Hello, I apologize for the very delayed response.

I'm having a bit of a trouble with installing the build you just sent. I went and downloaded the files from the commit, right clicked and installed the inf file from src>AmtPtpDeviceUniversalPkg and I get the error "The third-party INF does not contain digital signature information."

In the first I'm not exactly sure how to install the build so I'm not even sure if I'm doing it right.

imbushuo, 2020-05-17:
Please download the CI package instead of the commit itself. CI packages are signed.

---
Thanks,
Bingxing Wang

anuscleaner, 2020-05-17:
Pardon my ignorance, how exactly do I do that? Have been looking around trying to download it, the only way I've been able to download something so far was from GitHub, from the commit I mentioned. I'm not really sure what you mean by CI package. Tried to look around the site you sent and I wasn't able to find a way to download this CI package but the only thing I was able to download from it were the logs.

imbushuo, 2020-05-17:
Pick a recent commit with green check mark. Click the check mark, you will see a link to AzDevOps. Then download packages from “artifacts” section. Use the amd64 artifact: there’s an option to download all files as a zip package.

anuscleaner, 2020-05-17:
I clicked on the artifacts and apparently no feeds are found. Also tried searching for amd64 on the search box, still nothing.

imbushuo, 2020-05-17:
[cid:e79a45e9-13b6-4ef7-97e1-fef81a573850]
[cid:f4a7dd4c-2d15-4187-a9e1-b6a84e993b20]
---
Thanks,
Bingxing Wang

omscarr, 2020-05-24:
Hi there, I have a MacBookPro6,2, the track pad product id, vendor id, and phsical dimensions are the same as anuscleaner's above.

I tried installing from the artifacts of the build you posted by right clicking the AmtPtpDevice.inf file and clicking install.  But it didn't seem to have any effect on the trackpad.  How do we know if it has installed correctly and is working?  

Thanks again for your work

imbushuo, 2020-05-26:
If it is successfully installed, you should see a Precision Touchpad device under the HID category.

omscarr, 2020-07-01:
Ok, in that case it didn't install correctly.

I had another 3rd party touchpad driver installed [Trackpad++](http://trackpad.forbootcamp.org/) so I don't know if that was causing issues.  I'm not sure how to uninstall that first.

hannad4, 2020-08-09:
Yes, you should uninstall Trackpad++ first. Press Command key and R key. You should see a dialog box in the bottom left corner of the screen. Type in 'appwiz.cpl' and press enter. Locate Trackpad++ in the list of programs and uninstall it. Then perform the installation of this trackpad using the latest release available. Hope this helps you out.