imbushuo/mac-precision-touchpad #256: Two finger click with physical switch registers as regular left click

anohren, 2020-01-10:
**Pre-report checks**

- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
A right click fails to register about 50% of the time if you move the cursor with one finger, then stop moving but with the finger still on the touchpad add a second finger and press down in order to make a right click with the physical switch.

They register as single finger clicks, resulting in e.g. following a link instead of displaying its context menu.

**To Reproduce**
Steps to reproduce the behavior:
1. move cursor with middle finger
2. add ring finger to touchpad without lifting middle finger
3. press down to activate physical switch (MBA 2012 -- not tested with force touch)

**Expected behavior**
A right click. I.e. bring up the context menu in most cases.

**Traces**
-

**Environment**
Add your device model, driver type (USB or SPI) and OS version here.
Macbook Air 2012
USB most probably
Win 10 1910

Will double check next boot to Windows.

gautamjain, 2020-02-27:
I have the same issue on Win 10 1909 using Magic Trackpad 2 using Release 3805.

verity-s, 2020-07-01:
My understanding is that the right "click" only works if it's a "tap", not an actual click with trackpad feedback. I personally dislike all taps as that introduces false clicks way too many times, even at low sensitivity if the fingers happen to be resting on the trackpad. Is it true that we cannot disable two-finger tap and enable two-finger click only, to work as a right button? 

Same for three and four finger taps - have a choice to make them trackpad clicks?

anohren, 2020-07-01:
> My understanding is that the right "click" only works if it's a "tap"

That's not correct. It works both ways. 

> not an actual click with trackpad feedback. 

While there is feedback in the physical switch, there's no Taptic Engine in this model. 

> Is it true that we cannot disable two-finger tap and enable two-finger click only, to work as a right button? 
> 
> Same for three and four finger taps - have a choice to make them trackpad clicks?

Off topic for this issue, but it probably wouldn't be a good idea to only use multi finger clicks considering they're unreliable.

anohren, 2020-12-14:
Let me know if I can help with this in some way, @imbushuo.

qiukeren, 2022-03-18:
YES it treats right click as left click sometimes, and left click as right click some times

annoying