imbushuo/mac-precision-touchpad #257: Works Perfect With Magic Trackpad 2

DrJedikiah, 2020-01-10:
Firstly, this is not a bug report or an issue. I just wanted to say that I've been following this project for a year and tried several version. Right now I ditched the Magic Utilities software because this works exactly as it should and perfect. Now I can donate the budget that I prepared for Magic Utilities to you.

I have no sensitivity issues, no lagging at all.

Only things that I can say lacks of (not that important) are bluetooth support and force touch.
And since I use my mouse at 12.000DPI, I have disabled the acceleration in windows completely. Without it, even the fastest option in settings is not that fast that's just my idea, so I feel like it can be faster than this.

duggie1031, 2020-01-23:
Can I ask how you installed the files? I have looked at different ways but unable to work it out?

filipesilva, 2020-01-29:
In my case I had downloaded the drivers in the release page, and unzipped them. Then I connected the trackpad using the cable.

Windows just recognized and installed the device. It worked for a couple of seconds, then stopped working. I restarted the PC and it was working perfectly.

I could verify it was working precision drivers because I could use pinch when the windows trackpad settings had it on, and couldn't use pinch when I turned it off in the settings.

At first I thought Windows just had drivers for it already. I'm on the insiders slow ring ([this](https://blogs.windows.com/windowsexperience/2020/01/14/releasing-windows-10-insider-preview-build-19041-21-to-the-slow-ring/) build) and thought maybe they just added support there.

But then I went to the device manager -> human interface devices -> apple precision touchpad device, and saw that the manufacturer was listed as "Bingxing Wang", which is the same manufacturer as listed in the `AmtPtpDevice.inf` file in the release.

The driver in the device manager lists version 20.44.29.311 and data 28/02/2018 though, which isn't the same as my files on disk. So I pressed "Update Driver" on that screen and choose the ones on disk.

What I do know is that it works amazingly well. Amazing job @imbushuo! I just added a donation, your project enabled me to do something I just couldn't do otherwise.

duggie1031, 2020-01-29:
Thanks

I managed to get mine working with some help from Nick Wittwer on here - I think the main thing was I downloaded an older driver from the release page and this never worked but when downloading the latest one I was able to find the correct files to install 

Thanks for your response :)

imbushuo, 2020-01-29:
Yes I just corrected the CI procedure to eliminate confusing recently but forgot to mention it here.

imbushuo, 2020-01-29:
Bluetooth is a work-in-progress in early stage and I expect it to be available before April. For now, I am closing this issue and others will be tracked in separate issues.

closed 2020-01-29 by imbushuo