imbushuo/mac-precision-touchpad #258: My track pad stopped working on installing mac-precision-touchpad

mgupta11, 2020-01-13:
Hi 

Thank you for taking the effort to write this beautiful piece of code. I have a Macbook Pro 2015 and as soon as I installed mac-precision-pro my trackpad stopped working. Thankfully I had a blue tooth mouse of a friend for tonight so I am sailing thorough but would be in trouble tomorrow. 

Please see if you can help me uninstall in the meanwhile or check what the issue is and resolve it.

Thanks in advance
Manish

mgupta11, 2020-01-14:
Hi @reznikartem 

Sorry to see you vote down my comment. Would be grateful if you can share the reason as well. Am still struggling to find how to uninstall these drivers. Treid every thing mentioned on the forum but still no luck. Please help.

reznikartem, 2020-01-14:
@mgupta11 because of same issue at december. 
https://github.com/imbushuo/mac-precision-touchpad/issues/245

I think there is no reason to create duplicate.

mgupta11, 2020-01-15:
Oh ok.. My Bad.. I could not find the #245 Will check and update there.

imbushuo, 2020-01-16:
Could you attach a trace log of the driver bring-up?

benswinburne, 2020-04-10:
I had the same issue but went to uninstall following the instructions on the issue listed above and then reinstalled mac precision touchpad and it worked fine.

For what it's worth, I previously had Trackpad++ installed (and hadn't uninstalled it prior to installing this) so they perhaps conflicted?

imbushuo, 2020-06-04:
@benswinburne Make sure you've cleaned it up before installation.