imbushuo/mac-precision-touchpad #260: Blue screen of death after waking up laptop from sleep.

iceypotato, 2020-01-18:
**Pre-report checks**
- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
Similar to issue #203, but touchpad and keyboard do not work. Then after waiting for 1 minute or so (while the lid is still open), a BSOD appears with Stop code: DRIVER_POWER_STATE_FAILURE

**To Reproduce**
Close the laptop lid, then wake it from sleep using the Touchpad.

**Environment**
MacBook Pro TBT 2017 w/ Touch Bar using Windows 10 1903
Apple SPI Precision TouchPad version: 5.10.5.476
Using build 3768

imbushuo, 2020-01-19:
Do you still have C:\Windows\MEMORY.dmp retained? If you do have it, please share with the email address provided in my Github profile.

iceypotato, 2020-01-23:
Hi, I have sent you an email (niceawlego@gmail.com) containing my MEMORY.dmp file.

imbushuo, 2020-01-23:
Got it. I will take a look and follow up.

imbushuo, 2020-01-28:
```
STACK_TEXT:  
ffff8385`1bc3f778 fffff807`7acee3d7 : 00000000`0000009f 00000000`00000003 ffff9505`a1c67060 ffff8385`1bc3f7b0 : nt!KeBugCheckEx
ffff8385`1bc3f780 fffff807`7acee2e1 : ffff9505`a5e577f8 ffff9505`aa2a9230 ffffdd01`92680100 ffffdd01`92680180 : nt!PopIrpWatchdogBugcheck+0xef
ffff8385`1bc3f7f0 fffff807`7aa6ba89 : ffff9505`a5e57830 00000000`00000080 00000000`00000003 ffff9505`00000002 : nt!PopIrpWatchdog+0x31
ffff8385`1bc3f840 fffff807`7aa6a7e9 : 00000000`00000004 00000000`00989680 00000000`000b466d 00000000`00000085 : nt!KiProcessExpiredTimerList+0x169
ffff8385`1bc3f930 fffff807`7abc5024 : ffffffff`00000000 ffffdd01`92680180 ffffdd01`92691140 ffff9505`a89c2080 : nt!KiRetireDpcList+0x4e9
ffff8385`1bc3fb60 00000000`00000000 : ffff8385`1bc40000 ffff8385`1bc39000 00000000`00000000 00000000`00000000 : nt!KiIdleLoop+0x84
```

```
STACK_COMMAND:  .thread ; .cxr ; kb

FAILURE_BUCKET_ID:  0x9F_3_POWER_DOWN_IMAGE_AppleSPIDevice.sys

BUCKET_ID:  0x9F_3_POWER_DOWN_IMAGE_AppleSPIDevice.sys

PRIMARY_PROBLEM_CLASS:  0x9F_3_POWER_DOWN_IMAGE_AppleSPIDevice.sys

TARGET_TIME:  2020-01-18T05:50:10.000Z
```

This seemed to be the upstream issue I mentioned in the workaround. I haven't yet figured out the way to deal with this but will provide an update soon.

imbushuo, 2020-01-28:
Correlates with #203

imbushuo, 2021-05-30:
https://github.com/imbushuo/mac-precision-touchpad/releases/tag/2105-3965

closed 2021-05-30 by imbushuo