imbushuo/mac-precision-touchpad #261: Random mouse pointer movement

tigersfa8, 2020-01-22:
**Pre-report checks**

- [ ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Describe the bug**
When attempting to click on any spot on the screen, the mouse pointer jumps up and to the left approximately 1 mouse pointer size lengths 3 times, before staying on the spot and clicking what I have selected. Ex. I am about to click a weblink, but the mouse moves up and to the left approximately the distance of the length of my mouse pointer, when I push to click, or tap to click. It does this 3 time, and seems like it jumps up and to the left in the exact same position each of the 3 times.

This occurs when I'm browsing the web, navigating through windows explorer, different apps on my PC. 

**To Reproduce**
Random. About every hour or two hours, this issue occurs

**Expected behavior**
Click on the exact spot that I have placed the mouse pointer on to click on the screen.

**Traces**
n/a

**Environment**
Magic Trackpad 2, connected via USB to a port on my Keyboard, which is connected to a docking station for my desktop PC. Windows 10 1909 PC computer (not an Apple device)

reznikartem, 2020-01-22:
Have same issue too

imbushuo, 2020-01-23:
We are discussing about jitter elimination implementation with my friends, this should be done in the following months. I don't have a clear time frame at this moment though.

techopsadmin, 2020-01-28:
I have additional issues and wondering if I should open a new issue, or add to this one.
Additional issues:
1. Clicks are generated when fingers are resting on the trackpad. Ex. I'm moving the mouse around and I do not apply pressure, but randomly, the left click action is performed.
2. Drag is released without performing the action. This can relate to the issue above '1', but I am dragging an item (ex. file to a folder), holding down the left click with one finger, and using another finger to continue moving the file to the desired location. In the middle of the drag, the dragged item is released randomly.
3. When attempting to drag an item using the similar method as the above issue '2', the dragging feature stops and won't drag any further. Ex. I'm dragging a file to a different folder using one finger to hold down, and another finger to drag, but the dragging stops and I would have to start over.

Please let me know if I should create a new issue for the above.
Thank you,

cari66ean, 2020-03-29:
I have this issue too for several months now and I'd like to add my findings on this so far. 

So sometimes it happens repeatedly like 3 times or so in a row. Furthermore, it's mostly in the center of the trackpad (MT2) but that could be a coincidence, as I'm probably more likely to click in that area. 

However, what I'm pretty sure about by now is that it only happens with push to click... I've never noticed this with tap to click at all so far.

cari66ean, 2020-03-29:
Ok I think I found the issue here and I'm able to reproduce quite well. At first I noticed that it happens on certain movement patterns, e.g. right-click opening a context menu, moving the pointer down a little and clicking on the first button. It also happens very often when trying to repeat the same thing (probably doesn't make sense now but it has to do with the fact that you're getting quicker at the movement). 

Then it hit me... the pointer appeared to jump to the exact same spot it was before doing the small movement + clicking. I tried having a lookout and provoking it to happen a couple of times and indeed... it's always back to the same spot. 

After playing around some more I noticed how you can reproduce this basically at will as long as you're quick enough: 

- remove the finger from the surface
- touch the pad and swiftly move the pointer for a short distance or short time 
- click immediately without leaving the surface or stopping the movement before the click

Now I don't know what is more critical here... distance or time. But in order to trigger it, it has to happen very quickly after the movement has begun. The quicker you are the more likely the pointer will jump back. I can pretty much trigger it every time I'm focusing on it. 

This leads me to believe that it's not entirely a random bug per se but probably has to do with the jitter elimination implementation mentioned above by @imbushuo. It's purpose is probably to avoid accidental clicks when you move your finger on the pad while clicking and it would explain why I can't trigger it when tapping... you lift your finger before the tap, thus stopping any prior pointer movement anyway. Or maybe I'm just too slow due to having to lift my finger before the tap. 

Maybe you can release a test version without the jitter elimination on clicks (e.g. one that instead of triggering when you click during < certain distance or time limit one that is triggering after <0px or <0ms, so that it basically cannot be triggered and see what happens).

squalsoft, 2020-04-22:
This is very bad issue because click can happen anywhere on screen! I can test beta version on macbook pro 15" late 2013.

pd90506, 2020-05-23:
> Ok I think I found the issue here and I'm able to reproduce quite well. At first I noticed that it happens on certain movement patterns, e.g. right-click opening a context menu, moving the pointer [...]

I can confirm! (hard) Clicking after moving the cursor in a very short period of time will reproduce the issue.

imbushuo, 2020-05-24:
Sijie Bu (a friend of mine) is currently working on a fix for that.

omegamalkior, 2020-07-12:
> Ok I think I found the issue here and I'm able to reproduce quite well. At first I noticed that it happens on certain movement patterns, e.g. right-click opening a context menu, moving the pointer [...]

This is really accurate! Kudos

> Sijie Bu (a friend of mine) is currently working on a fix for that. Get Outlook for iOS<https://aka.ms/o0ukef>

Sounds heavenly 🙏

cari66ean, 2020-09-10:
Just installed the new release and if I'm not mistaken... I cannot reproduce this anymore. :)

Cyberdroid1, 2021-01-05:
Any updates to this?

jjoshrtca, 2021-06-16:
I have this issue, too.  I think it's related to another issue.  I just posted to https://github.com/imbushuo/mac-precision-touchpad/issues/375

I think it has something to do with the slight fog/condensation that appears on the trackpad around the finger/s (or under the fingers when slightly hovering).

Thanks @cari66ean for letting us know that this issue does not generally happen with taps!  Looks like I may have to learn to live with tapping 🤢

sanarena, 2022-08-13:
I am experiencing same issue too. would appreciate if someone can provide a fix or remedy.