imbushuo/mac-precision-touchpad #262: Palm Rejection

kokowang, 2020-01-25:
**Pre-report checks**
- [x] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Is your feature request related to a problem? Please describe.**
Using the newer Macbooks without palm rejection is difficult due to the size and position of the trackpad. 

**Describe the solution you'd like**
Implement global palm rejection.

**Describe alternatives you've considered**
Disabling the trackpad while typing is not ideal. I use the laptop for software development and not being able to move the cursor for a short time after editing is bothersome. This also does not fix using the trackpad with your palm on it (detects as 2 finger scroll). 

**Additional context**
Macbook Pro 15" 2019

sukov, 2020-03-22:
Same here, in my opinion this is the only important missing feature and it is kind of a deal breaker. Other than that, I really think that this is the best driver for the Trackpad!

Any info on a road-map for this enhancement? Thanks!

Anatharias, 2020-04-06:
Hello,
Same for me. the MBP 16 has a Trackpad that is so big, that palm rejection is required for proper interaction. but when used from the side, it offers a greater experience than Apple Trackpad driver.
Thank you

yiw02, 2020-04-23:
same issue here, I feel that the only issue is the palm rejection is not present.

JeroenGar, 2020-04-27:
I have the same issue. Using the Magic Trackpad 2 on my desktop. It's just too big avoid accidentally touching the trackpad.

hranum, 2020-06-01:
Same issue here, the trackpad is so large and it becomes unusable when slouching in the sofa without external mouse/kb, because I accidentally come in contact with the palm area of the touchpad all the time.

yulei-xu, 2020-07-06:
Same issue, it makes using the MBP 16 a bit difficult. If implemented, this would make the MBP be one of the best Windows laptops out there, funnily enough.

omegamalkior, 2020-07-12:
> Try this.
> https://docs.microsoft.com/en-us/windows-hardware/design/component-guidelines/touchpad-tuning-guidelines
> 
> I had luck with `SuperCurtainLeft` `SuperCurtainRight` and `SuperCurtainTop` to 2000 himetric units.

This did not work for me. Did not see this in Registry and when I manually added it in it did nothing

kjavia, 2020-07-13:
@omegamalkior you need to reboot after adding the 3 new registry entries. I will add a screenshot later. It definitely disables the areas of the touchpad (if you start touching from those areas). The areas still receive touch if the touch is originated from outside those areas.

omegamalkior, 2020-07-13:
> @omegamalkior you need to reboot after adding the 3 new registry entries. I will add a screenshot later. It definitely disables the areas of the touchpad (if you start touching from those areas). The areas still receive touch if the touch is originated from outside those areas.

If you can upload the screenshots I would be grateful as yeah even if I do originate the touchpad from those areas they still happen

kjavia, 2020-07-14:
[image]

rolodato, 2020-07-14:
You can save this as a .reg file to make the edits quicker:

```
Windows Registry Editor Version 5.00

[HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\CurrentVersion\PrecisionTouchPad]
"SuperCurtainLeft"=dword:00000bb8
"SuperCurtainRight"=dword:00000bb8
"SuperCurtainTop"=dword:000003e8
```

phaseOne, 2021-08-20:
Anyone know of a way to test changes to these parameters without rebooting?

On a related topic, the Microsoft docs mention that the above tuning values are in himetric units (1 himetric = 10 µm = 0.01 mm). I'm wondering how the physical size of the trackpad gets defined in this driver. Are the units in the `SpiTrackpadConfigTable` in `AppleDefinition.h` physical himetric units or logical units?