imbushuo/mac-precision-touchpad #264: Can't turn Secure Boot back on...

slminneman, 2020-02-03:
So, I successfully installed the drivers on my MBP 16" running Windows 10 Pro.

As suggested, I had to turn Secure Boot off.  I was using old instructions and did the old TESTSIGNING step.

The drivers work fine, and I love all of the new functionality, but I can't turn Secure Boot back on.  If I select high security, it tell me I need to have network on for some reason, but there's no apparent way to do that.

If I select medium security, it'll allow that choice, and the subsequent boot has no trackpad pointer (if I Command-R and select no boot security, it comes back).

Any suggestions?  ...or is this expected behavior?

Thanks!  Scott

imbushuo, 2020-02-04:
You need to turn off Secure Boot at now, but `TestSigning` is not necessary. You can let Windows to verify signatures and this will go through fine.

closed 2020-02-04 by imbushuo

slminneman, 2020-02-04:
But, like I said, the machine will not let me turn Secure Boot back on.  Do I have to leave it off permanently?

 

Thanks!

 

Scott

imbushuo, 2020-02-04:
This will last for a while (at least a few more months), then the situation will be re-evaluated.

Thanks,
Bingxing (Ben) Wang