imbushuo/mac-precision-touchpad #268: Cursor randomly stops moving

ItsPops, 2020-02-24:
Hello,
I've been using successfully the driver on my 2017 15" MBP. Thanks for your work.
I now have a 16", i9 2.3GHz, SPI trackpad (I guess), Windows 10 1909

Sometimes the cursor stops moving (when doing small circular movements for example) then start to move again from the place it stopped. I don't have to release the finger to make the cursor to move again. Not only happening in the dead zones, but also when moving finger at the center of the trackpad.

geirsagberg, 2020-03-01:
I notice this happens when I use the pointy tip of my finger; presumably because there is thicker skin there, with less conductivity? A workaround is using the flat of the finger instead.

morty89, 2020-03-05:
Like geirsagberg, this also happens to me when i'm using the tip of my finger. When pressing harder or using my whole finger-surface, it's fine. This is an issue as i can't relax as i have to press harder to use the trackpad.

In simple words. the trackpad sensitivity is too low. Waay lower than in MacOS, i checked.
I'm using a mid-2012 MBP 15", 2.6ghz, GT650m.

kevinm067, 2020-03-05:
Same thing happens to me, when I am using macOS the trackpad works smoothly. I have to press harder while using windows. I hope the dev fixes this issue

BasO, 2020-03-11:
Same issue, on 2019 MacBook 13". Makes it unviable for me.

squalsoft, 2020-04-22:
Yes! Very annoying issue.