imbushuo/mac-precision-touchpad #270: A way to adjust trackpad sensitivity

hanseytrackpad, 2020-02-28:
I just installed the drivers and I am supper happy about the result, this owns Trackpad++ drivers. However, I have super slow trackpad mouse pointer speed. Is there a way or will there a way to adjust the trackpad mouse pointer speed? Right now I am also using a logitech g403 mouse because the trackpad is way to slow for me.

hanseytrackpad, 2020-02-28:
Okay I found the settings with Win + i , hotkey and then trackpad. However even at 10 in speed it is too slow in traversing the 2880x1800 Macbook Pro 15" Retina late 2013 resolution.
Does anyone have a solution?

hanseytrackpad, 2020-02-28:
Okay, I found a fix. I installed the program in the link and set the x and y values as in the first post.
I will not be going back here to check for a response any time soon, but if others have the same problem this is how I solved it. 
https://www.esreality.com/post/2363191/custom-mouse-accel-program-for-windows/

b-hayes, 2020-05-23:
Yeah, this def needs addressing. Pointer speed is sooooo slow.. I have to choose between functional gestures and pointer speed. Forgot about the workaround and searched for this issue again.