imbushuo/mac-precision-touchpad #272: Two-finger back swipe

Xardas220, 2020-03-09:
Hi, 
is there a chance to add the feature that macs have to swipe with two fingers (like right click) to the right to go back for example in the browser? Or the opposite to swipe left with two fingers to go forward.

I use a Trackpad 2 on regular Windows 10 with USB -> Lightning.

Thank you for your fantastic work, 

Leon

lcharles, 2020-03-10:
You should try to look at the touchpad settings in windows 10, because it works for me with my magic trackpad 2.

Jamesits, 2020-03-10:
I think this depends on application support. I'm also on Windows 10 ans using MT2, the gesture works on Chrome but not on Firefox.

Xardas220, 2020-03-10:
Thanks but I know there are more options. But there are only 3-finger and 4-finger gesture-options for me.
I think it's a problem with Vivaldi - the browser I am using. I going to look in the options for that.

imbushuo, 2020-03-13:
Chrome (and the new) Edge have two finger swipe. The support depends on application, which is out of my control.

closed 2020-03-13 by imbushuo

snowe2010, 2020-07-19:
@imbushuo what is the feature that we need to push Mozilla to implement for this to work?

imbushuo, 2020-07-19:
@snowe2010 Yeah, FF guys never implement that....

snowe2010, 2020-07-19:
Sorry I was asking what we need to request for them to fix. I will create a bug report, but do not know how to word it.

imbushuo, 2020-07-19:
@snowe2010 Just ask them if they have plan for implement the technology called "Direct Manipulation" on Windows 8+ platform. It is essentially the API set for Windows touch devices including touch screen, precision touchpad and other pen inputs.

snowe2010, 2020-07-19:
Awesome. Will do. Thanks!

snowe2010, 2020-07-19:
Looks like there is [an existing issue] that is being worked on. There is a flag, `apz.windows.use_direct_manipulation`, but it does not do anything for 2 finger swipe currently.

lyscop, 2020-08-29:
> Chrome (and the new) Edge have two finger swipe. The support depends on the application, which is out of my control.

I find this function in Chrome and Edge, and I disable the Overscroll history navigation option in edge://flags/ , now I want to set the Last tab/Next tab shortcuts of 2 fingers swipe left/right, and there is no option to do this.

Is there no 2 finger swipe left/right setting?