imbushuo/mac-precision-touchpad #275: Bluetooth Support

ericdhkim, 2020-03-14:
Hi,

New here so please forgive my ignorance if I missed something. I just got a Apple Magic Trackpad 2 to work with my Windows 10 PC. I've downloaded the latest release from here and installed the .inf file. When I plug it in via USB, Windows 10 does recognize it as a precision trackpad, but when I connect via Bluetooth, I lose the "touchpad" option in settings altogether leading me to believe that over Bluetooth, it's just recognized as a mouse. I see on the roadmap that bluetooth is unchecked, does this indicate that what I've described is operating as it should, until bluetooth support is implemented, or that I've done something wrong?

Thanks,

AhmadChaiban, 2020-03-15:
Hi!
This is happening on my end too. It’s not recognized as a precision touchpad. I don’t think these drivers support Bluetooth yet and that it’s operating as it should.

jespertheend, 2020-03-16:
https://github.com/imbushuo/mac-precision-touchpad/issues/229
Bluetooth support is expected before the end of this quarter. I'm using usb for the time being.

imbushuo, 2020-03-17:
Duplicate of #229. By the way it is coming very soon...

closed 2020-03-17 by imbushuo