imbushuo/mac-precision-touchpad #276: Tap to click not always detected

alanwww1, 2020-03-16:
For a 2013 Macbook pro, everything works fine except, at the 10% of my tap to clicks does not get recognised.  Is there a way to change senisbilty in this direction. Where can I find a hardcoded value in the source code ? I tried to change the windows sensitivity setting, but no luck with that.

Thanks,
Attila

Lambozhuang, 2020-03-27:
MacBook Pro 2019 4TBT3  has the same issue. It won't detect all the tap to clicks , which is very annoyed. Please fix this bug soon. Thank you!!

DrBineshian, 2020-05-14:
I am using MacBook Pro 2019. Tap to click using the latest release does not register for at least 50% of the cases. I believe this is a bug to be fixed. Please kindly look after the issue.

JayDoubleu, 2020-11-20:
Looks like it's not registering the "fast" taps. If you tap and hold for millisecond then it works every time. I'm guessing it's something to do with sensitivity to register the taps itself. This is crucial for me so I'll revert to bootcamp assistant drivers until fixed.