imbushuo/mac-precision-touchpad #277: Disable right click with two fingers pressing.

squalsoft, 2020-03-19:
**Pre-report checks**
- [x ] Have you checked (use search) [Azure DevOps Board](https://ligstd.visualstudio.com/Apple%20PTP%20Trackpad/_workitems/recentlyupdated) and Github issues for known bugs and plans?

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Frequently when i click immediatelly after moving cursor i receive right click action.

squalsoft, 2020-03-26:
By the way it exists some setting (in registry maybe) to disable right click on touchpad?

squalsoft, 2020-03-26:
I found solution with AutoHotkey: 
RButton::LButton

plainerman, 2020-04-08:
Assuming you are on Windows 10, you can change it in the normal precision driver settings.
Settings > Devices > Touchpad >[Taps] Tap with two fingers to right-click.

@imbushuo I believe this issue can be closed.

squalsoft, 2020-04-08:
I talk about clicks not taps. So this can't be disabled in windows. But i close this issue because this problem is much deeper.

closed 2020-04-08 by squalsoft